Klipper3d/klipper #2090: Crush on BLTouch clone `PROBE` - debug commands work

dzervas, 2019-10-20:
[klipper.log](https://github.com/KevinOConnor/klipper/files/3747906/klipper.log)

I have a BLTouch clone (I know how it should behave, I have the same sensor on an Ender 3 with Marlin) attached on a delta printer (almost Kossel - mixed design with rods).

`BLTOUCH_DEBUG` commands work as expected, apart from `touch_mode`. It does nothing. Is that normal?

Apart from that, the main problem is that when I give `PROBE` command or other probing commands (DELTA_CALIBRATE, etc.), klipper crushes (see log for more):

> ERROR:root:Unhandled exception during run
> Traceback (most recent call last):
>   File "klippy/klippy.py", line 173, in run
>     self.reactor.run()
>   File "/home/dzervas/Documents/Rossel/klipper/klippy/reactor.py", line 250, in run
>     g_next.switch()
>   File "/home/dzervas/Documents/Rossel/klipper/klippy/reactor.py", line 277, in _dispatch_loop
>     timeout = self._check_timers(eventtime)
>   File "/home/dzervas/Documents/Rossel/klipper/klippy/reactor.py", line 135, in _check_timers
>     t.waketime = waketime = t.callback(eventtime)
>   File "/home/dzervas/Documents/Rossel/klipper/klippy/reactor.py", line 47, in invoke
>     res = self.callback(eventtime)
>   File "/home/dzervas/Documents/Rossel/klipper/klippy/mcu.py", line 210, in _home_retry
>     self._trigger_notify()
>   File "/home/dzervas/Documents/Rossel/klipper/klippy/homing.py", line 37, in _endstop_notify
>     self.toolhead.signal_drip_mode_end()
>   File "/home/dzervas/Documents/Rossel/klipper/klippy/toolhead.py", line 461, in signal_drip_mode_end
>     self.drip_completion.complete(True)
> AttributeError: 'NoneType' object has no attribute 'complete'

KevinOConnor, 2019-10-23:
It appears the attach log file is not complete.  We need the full log in order to assist.

-Kevin

dzervas, 2019-10-23:
I made `pin_up_reports_not_triggered` and `pin_up_touch_mode_reports_triggered` `False` and it seems to be working. I don't have a consistent behavior, so I'll close till I gather more data (and a complete log)

closed 2019-10-23 by dzervas