Klipper3d/klipper #2091: Configuration for SKR 1.3 and CR-20 Display UC1701

arcticlinux, 2019-10-20:
- Creality CR-20
- Creality ABL Bed Levelling on Z Endstop
- BigtreeTech SKR1.3
- 4x TMC2130

Just trying to connect the default Creality CR-20 display to the SKR 1.3 in EXP1 and EXP2 not sure how to do the configuration.  Tried searching through previous issues.  Best I came up with was that the SPI bus is shared between the TMC2130s and the UC1701 display.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3748129/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3748130/printer.cfg.txt)

arcticlinux, 2019-10-20:
The panel wasn't lighting up, various discussions about inverted cables.  After reviewing the generic 12864 panel documentation (which shows the VCC/GND on pins 1/2) I cut off cable key and reversed the EXP1 cable, which makes the LCD display light up.

https://wiki.fysetc.com/Generic_12864_Panel/

That didn't make the display work, so additional configuration is probably required, not sure if I'm heading down the right path.

KevinOConnor, 2019-10-23:
Are you sure it's a "fysetc 12864" type display as opposed to a "mks mini 12864" display?

FYI, the skr does not share spi pins with the stepper drivers - the stepper drivers have their own (software spi) pins.

-Kevin

arcticlinux, 2019-10-23:
Sure, it's more like the Mini, the pinout is the same. https://wiki.fysetc.com/Mini12864_Panel/#typical-wiring

I'll give it another try, just haven't been able to get anything to display.

arcticlinux, 2019-10-24:
Removed the spi config lines, used the standard UC1701 config, and had to flip over both cables and all is working.

Thanks for your work and responding to issues!

closed 2019-10-24 by arcticlinux