Klipper3d/klipper #2092: How do i enable virtual sdcard?

shubble32, 2019-10-20:
i am having issues trying to enable virtual sdcard feature. i am running klipper on my ender 3 can someone please explain this to me? Thanks

BlackStump, 2019-10-20:
have you read these sections?
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L444

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#g-code-sd-card-commands

closed 2019-10-27 by klipper-gitissuebot

Snobs76, 2020-03-19:
I`v got alwys a M112 errot if i enable these two line
[virtual_sdcard]
path: ~/.octoprint/uploads/

I cant figure it out why this happens. Printer did not work anymore :(

Any Idea for me.

ld3300, 2020-03-20:
This is why the issue is requesting a log file.  Please follow the instructions provided above by klipper-gitissuebot so we can see what the setup is and help contribute to a solution.

Snobs76, 2020-03-30:
Sorry for the late answer, I have tried something around and figuered it out. I used an "old" version of klipper mixed up in aother directory with a newer on. It starts, now with active VIRTUAL_SD. I have to try it out. Hopefully are my problems with stutter printering are gone