Klipper3d/klipper #2094: How to resolve gcode_id to heater name in a macro, or through SET_HEATER_TEMPERATURE command

simonkuehling, 2019-10-21:
Hi,

I am trying to extend a custom printer host software to automatically discover all available heaters that are currently configured in Klipper - in order to offer temperature controls in the interface for each of them.
It is my goal to not have to manually configure this knowledge in the host - from a "don't repeat yourself" standpoint.
So far I can easily gather all heaters by their gcode_ids that are delivered with M105 temperature reports - but it is currently not possible to control them through the SET_HEATER_TEMPERATURE command, since the host does not have access the the gcode_id <--> heater name mapping right now.

Example:
* configure a generic_heater named "chamber_heater"
* assign gcode_id "C"

Problem:
* resolving "C" to "chamber_heater" on the host not available

I see several options to add this capability to Klipper, which I would like to implement and PR, but I am not sure to decide which one is the best route to go - so I would love to get some feedback on this:

1) Implement a new command like "SET_HEATER_TEMPERATURE_BY_ID ID=<gcode_id>" that works very much like SET_HEATER_TEMPERATURE right now - just resolves the gcode_id to heater name somehow.

2) push the gcode_id <--> heater name mapping from PrinterHeaters into the gcode macro environment through a get_status method in PrinterHeaters, maybe?. This way I could use a custom macro to implement the desired functionality

3) extend the SET_HEATER_TEMPERATURE command to also accept the gcode_id as an alias name for the HEATER parameter - although I have already seen that the gcode_id parameter is not available in the scope of a Heater object where cmd_SET_HEATER_TEMPERATURE() is living.

(empty log) [klippy.log](https://github.com/KevinOConnor/klipper/files/3750029/klippy.log)

KevinOConnor, 2019-10-23:
It's hard to make recommendations about future code, but off the top of my head, I'd say option 2 sounds the most appealing.

-Kevin

simonkuehling, 2019-10-25:
Hi Kevin, thanks for weighing in - I'll try to implement it that way.

With that I am running into a problem though: My first guess was adding a get_status() method to the PrinterHeaters class like this (just a static response object for testing right now):
```
get_status():
    return { 'list': {"T0": "extruder0", "B": "heater_bed"}}
```
expecting to then be able to use ```printer.heater.list``` in a macro. But I get an error:
```
!! Error evaluating 'gcode_macro HEATERS:gcode': UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'heater' 
```
From the code I thought that the internal printer.heater is an instance of PrinterHeaters - and with a get_status() it should be accessible through GetStatusWrapper. Am I missing something here?

KevinOConnor, 2019-10-30:
As far as I know, it should have worked.  It's hard to say from just a snippet of code.

-Kevin

simonkuehling, 2019-11-07:
Ok, got it working - I forgot to add the two expected parameters to the get_status() function.
This is working now, I will add the actual aggregation of the list now and will make a PR for your review when ready.

```
get_status(self, eventtime):
    return { 'list': {"T0": "extruder0", "B": "heater_bed"}}
```

KevinOConnor, 2019-12-16:
Is this resolved with the merge of #2159?

-Kevin

simonkuehling, 2019-12-16:
Yes, that is resolved now - closing this issue!
Simon

closed 2019-12-16 by simonkuehling