Klipper3d/klipper #2096: Feature request: support [first_layer_height] macro in Gcode

klcjr89, 2019-10-21:
Hello, I discovered that my S3D start script doesn't work in Klipper. My start script is customized to prime and purge the nozzle off the edge of the build plate and then sets the nozzle position to the first layer height accordingly.

S3D has a [first_layer_height] macro that I used to use on another firmware that worked.

As a temporary workaround, I had to manually set this to be Z0.2, as most of my first layer's are typically this height.

However, I would rather it be dynamic the way it used to be, as I do want to venture into trying different first layer heights depending on the model.

It's also worth noting that Slic3r also seems to have this macro available based on my research.

Below is my start gcode script that makes use of the S3D [first_layer_height] macro:

`M107 ; Layer fan off
M104 S200 ; Set hotend temperature with no wait
M190 S[bed0_temperature] ; Set bed temperature and wait
M109 S200 ; Set hotend temperature and wait
G28 ; Home all axes
BED_TILT_CALIBRATE ; 3 point bed tilt
G1 X-10 Y-2 F7200 ; Move nozzle to the priming location
M400 ; Wait for current moves to finish
G1 Z-0.3 ; Move nozzle below the bed
M109 S[extruder0_temperature] ; Set hotend temperature and wait
G92 E0 ; Reset extrusion distance
G1 E20 F400 ; Extrude 20mm of filament
G92 E0 ; Reset extrusion distance
G1 E-0.6 F3600 ; Retract
G92 E0 ; Reset extrusion distance
G1 E0.59 F400 ; Reprime
G92 E0 ; Reset extrusion distance
G1 X30 F400 ; Move nozzle across right finger tab to remove purge
M400 ; Wait for current moves to finish
G1 Z[first_layer_height] ; Move Z to the first layer height
M117 Printing ; Display Printing message`

klcjr89, 2019-10-21:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3752624/klippy.log)

Log attached to satisfy github bot

FHeilmann, 2019-10-22:
`[first_layer_height]` is an S3D variable that S3D is supposed to replace with `0.2` during the slicing process. The string `[first_layer_height]` is never supposed to remain in the gcode. There is no way for klipper to handle this and I'd wager that no other firmware handles this either. Check your start script for typos and verify that the resulting gcode actually has `0.2` inserted instead of `[first_layer_height]`.

klcjr89, 2019-10-22:
@FHeilmann There aren't any typos in the script, but it's also worth noting the resultant gcode generated doesn't replace the macro for `G` Z0.2`, it still shows: `G1 Z[first_layer_height]`. This worked fine on Marlin, so it's not an issue with my script. Looking for a workaround until/if it's implemented in the main branch via my cfg file, as I asked in the original post.

FHeilmann, 2019-10-22:
I can definitely tell you that marlin does not know how to handle [first_layer_height] in gcode (if you don't believe me you can search marlins code base for this string, you won't find any references to it in the code). The start script you were using with marlin likely moved the Z position correctly which effectively masked this issue.

I had another look at the list of g-code placeholders simplify3d provides, (see here: https://forum.simplify3d.com/viewtopic.php?f=8&t=1959&p=7574) [first_layer_height] is not one of them which is why it doesn't get expanded by S3D.

HTH

Florian

Hywelmartin, 2019-10-22:
@klcjr89  It's a problem with your script. if the gcode works with Marlin it is due to that Marlin doesn't care of bad gcode..

jakep82, 2019-10-22:
I will just reiterate what @FHeilmann said.  If you look at your script you'll see other variables such as [bed0_temperature] and [extruder0_temperature].  These are replaced with actual numbers by S3D during slicing.  That isn't happening with your first layer line, and Marlin is probably simply ignoring that line.

klcjr89, 2019-10-22:
That doesn't explain how I can solve it, which is what I was asking in the original post. Using different layer heights worked on Marlin using the script but I prefer to use Klipper.

klcjr89, 2019-10-22:
That doesn't help at all. Why not explain how to fix it then?

Hywelmartin, 2019-10-22:
M107 ; Layer fan off
M104 S200 ; Set hotend temperature with no wait
M190 S[bed0_temperature] ; Set bed temperature and wait
M109 S200 ; Set hotend temperature and wait
G28 ; Home all axes
BED_TILT_CALIBRATE ; 3 point bed tilt
G1 X-10 Y-2 F7200 ; Move nozzle to the priming location
M400 ; Wait for current moves to finish
G1 Z-0.3 ; Move nozzle below the bed
M109 S[extruder0_temperature] ; Set hotend temperature and wait
G92 E0 ; Reset extrusion distance
G1 E20 F400 ; Extrude 20mm of filament
G92 E0 ; Reset extrusion distance
G1 E-0.6 F3600 ; Retract
G92 E0 ; Reset extrusion distance
G1 E0.59 F400 ; Reprime
G92 E0 ; Reset extrusion distance
G1 X30 F400 ; Move nozzle across the bed to remove purge
M400 ; Wait for current moves to finish
M117 Printing ; Display Printing message

klcjr89, 2019-10-22:
@Hywelmartin Close but not quite, the Z needs to move up to the correct first layer height before the XY moves occur.

jakep82, 2019-10-22:
Again, you are asking for Simplify3D support on the Klipper Github.  You should post this question on the S3D forums.

klcjr89, 2019-10-22:
Sorry but that is not what I'm doing here; I have moved away from a working Marlin config and am trying to implement the lost functionality whether it means changing my .cfg file to use a macro, or to modify the code. Maybe Marlin doesn't care about 'bad gcode' but nonetheless it worked with no explanation as to why, because it's not apparent in their gcode parser source.

jakep82, 2019-10-22:
Three people have said the same thing.  You are using an invalid variable in your S3D start script.  This cannot be solved by Klipper or any other firmware.  I would suggest simply deleting the bad line as @Hywelmartin recommended.

FHeilmann, 2019-10-22:
Let me formulate a hypothesis on why you didn't see any issues:

- Your starting script homes Z via G28. 
- Most likely your nozzle retracts after triggering the Z endstop.
- You command your printer to move to Z-0.3 Now marlin does not allow negative Z positions unless you explicitly configure it so, so I'm guessing it either moved to Z0 and stopped there, or did not move to from whatever your retracted Z height at all.
- Marlin throws out the Gcode with the unexpanded S3D parameter because it's not valid gcode
- Marlin executes the XY move.
- S3D has the habit of explicitly moving the nozzle to the starting Z height before starting to extrude, so your nozzle ends up at your desired Z height.

Finally I'd like to remind you that we are all volunteers here. So I suggest you adopt a bit of a friendlier tone if you want people to continue helping you. We don't owe you any help.

klcjr89, 2019-10-22:
@FHeilmann I always configure my firmware to allow negative Z values.

It's funny that you mention how to be treated and for me to act more friendly (which my first post was completely poilite asking for a feature request, not an actual reported issue) when a majority of people in this thread have treated me in a condescending way.

closed 2019-10-22 by klcjr89