Klipper3d/klipper #2097: LCD uc1701 rotate display

dharvey242, 2019-10-22:
CETUS MK2 with Tinyfab CPU

Can anybody shed any light on how to rotate the display on the uc1701 (viki2 lcd).  It works great albeit upside down, config is currently:
[display]
lcd_type: uc1701
cs_pin: P0.16
a0_pin: P2.6
encoder_pins: ^!P3.25, ^!P3.26
click_pin: ^!P2.11
contrast: 40
menu_root:

smoothieware config (worked) was:
# Example viki2 config for Re-ARM with IDC cable
panel.lcd                                    viki2             # set type of panel
panel.spi_channel                            0                 # set spi channel to use P0_18,P0_15 MOSI,SCLK
panel.spi_cs_pin                             0.16              # set spi chip select
panel.spi_frequency                          200000
panel.encoder_a_pin                          3.26!^            # encoder pin
panel.encoder_b_pin                          3.25!^            # encoder pin
panel.click_button_pin                       2.11!^            # click button
panel.a0_pin                                 2.6               # st7565 needs an a0
panel.contrast                               4                 # override contrast setting (default is 9)
panel.encoder_resolution                     4                 # override number of clicks to move 1 item (default is 4)
panel.pause_button_enable                    true
#panel.button_pause_pin                      1.22!^             # kill/pause set one of these for the auxilliary button on viki2
panel.back_button_pin                       1.22!^            # back button recommended to use this on EXP1
panel.buzz_pin                               0.25              # pin for buzzer on EXP2
panel.red_led_pin                            2.8               # pin for red led on viki2 on EXP1
panel.blue_led_pin                           4.29              # pin for blue led on viki2 on EXP1
panel.external_sd                            true              # set to true if there is an extrernal sdcard on the panel
panel.external_sd.spi_channel                0                 # set spi channel the sdcard is on
panel.external_sd.spi_cs_pin                 1.23              # set spi chip select for the sdcard
panel.external_sd.sdcd_pin                   1.31!^            # sd detect signal (set to nc if no sdcard detect)
panel.menu_offset                            0                 # some panels will need 1 here
panel.reverse                                false              # reverse the lcd

dharvey242, 2019-10-22:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3754600/klippy.log)

KevinOConnor, 2019-10-23:
The software doesn't currently support screen rotation.  It's certainly possible to do that, but a developer would need to implement it.

-Kevin

KevinOConnor, 2019-11-06:
I'm going to mark this as inactive as it doesn't look like a developer is currently working on it.

-Kevin

closed 2019-11-06 by KevinOConnor