Klipper3d/klipper #2099: Unknown spi_bus '<board>:spi'

vladbabii, 2019-10-22:
I'm building a multi-board printer (right now one (board mcu)mega2560 and two minis (board xy and z), will add two more minis in the near future) with:
- 2x 5160 on X board
- 3x 2209 on Y board

The 2209 with uart work fine with <board>:pin for uart but the <board>:spi does not seem to work. I tried x:spi and x:spi1.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3757026/klippy.zip)

Please advise how to configure SPI in this case.

Thank you

jakep82, 2019-10-22:
I don't think you need to specify the bus unless you aren't using the main SPI bus.  It might work if you simply remove that line, but I'm not certain.  Also, you should look at your cs_pin for the X stepper.  You've specified an Arduino pin alias which won't work on that MCU the way you have it set up.

vladbabii, 2019-10-22:
I have tried
- software pins on board 2 - multiple pins, including the hardware ones
- hardware spi on arduino (board 1)

So is the only way to have spi on main board? I tried using spi and connecting the stepper to pins 50 and 51 then connecting the cs pin to pin 22 but no result.

I was trying to use spi on second board, where the step sticks are. If only the first board can have spi then that limits me for the next 10+ steppers - i can only order uart ones, 2208 and 2209.

vladbabii, 2019-10-22:
board 1 (arduino mega - no ramps, just a mega)
  - nothing connected

board 2 (skr mini 1.1)
 - X: 5160 with spi for X
 - Y: 5160 with spi for Y
 - Z: 2208 standalone (just for testing, i know it does not work for normal use)
 - X endstop for X endstop
 - Y endstop for Y endstop

board 3 (skr mini 1.1)
  - X: 2209 with uart for Z1
  - Y: 2209 with uart for Z2
  - Z: : 2209 with uart for Z3

I need another 10 steppers sockets for extruders (i'm doing a corexy toolchanger). What would be the supported solution now?

jakep82, 2019-10-22:
I'm not sure what you're actually trying to accomplish, but you can't split steppers on the same axis between different MCUs.  If you need more than one stepper for an axis, they all need to be on the same MCU along with the endstop for that axis.

If you want to use software SPI pins on the SKR mini boards, you need to specify those pins and not use the spi_bus parameter.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L1474

vladbabii, 2019-10-22:
They are on the same axis. Above i just put socket first,
- no steppers on board 1
- X and Y steppers are on board 2
- Z, z1 and z2 steppers are on board 3

vladbabii, 2019-10-22:
board 1 (mega spi)
```
[tmc5160 stepper_x]
cs_pin: ar32 #xy:PC13
spi_bus: spi
microsteps: 16
interpolate: True
run_current: 0.9
stealthchop_threshold: 100
```
result of dump

```
Recv: // ========== Write-only registers ==========
Recv: // TPWMTHRS:   00000025 TPWMTHRS=37
Recv: // COOLCONF:   00000000
Recv: // IHOLD_IRUN: 00060808 IHOLD=8 IRUN=8 IHOLDDELAY=6
Recv: // TPOWERDOWN: 0000000a TPOWERDOWN=10
Recv: // ========== Queried registers ==========
Recv: // GCONF:      00000000
Recv: // CHOPCONF:   00000000 MRES=0(256usteps)
Recv: // GSTAT:      00000000
Recv: // DRV_STATUS: 00000000
Recv: // FACTORY_CONF: 00000000
Recv: // IOIN:       00000000 VERSION=0x0
Recv: // LOST_STEPS: 00000000
Recv: // MSCNT:      00000000
Recv: // MSCURACT:   00000000
Recv: // OTP_READ:   00000000
Recv: // PWMCONF:    00000000
Recv: // PWM_SCALE:  00000000
Recv: // PWM_AUTO:   00000000
Recv: // TSTEP:      00000000
```

next: software spi on board 2

jakep82, 2019-10-22:
That dump means SPI is not working.  You can't split the stepper like that.  Your step, direction, and enable pins need to be on the same board as your SPI pins.

vladbabii, 2019-10-22:
I am just trying all options 

```
[stepper_x]
step_pin: xy:PC6
dir_pin: !xy:PC7
enable_pin: !xy:PB15
step_distance: 0.00498118
endstop_pin: ^!xy:PC2
position_endstop: 0
position_max: 450
homing_speed: 10

[tmc5160 stepper_x]
cs_pin: xy:PC13
spi_software_sclk_pin: xy:PB7
spi_software_mosi_pin: xy:PB6
spi_software_miso_pin: xy:PC12
microsteps: 16
interpolate: True
run_current: 0.9
stealthchop_threshold: 100
```

```
Send: DUMP_TMC stepper=stepper_x
Recv: // ========== Write-only registers ==========
Recv: // TPWMTHRS:   00000025 TPWMTHRS=37
Recv: // COOLCONF:   00000000
Recv: // IHOLD_IRUN: 00060808 IHOLD=8 IRUN=8 IHOLDDELAY=6
Recv: // TPOWERDOWN: 0000000a TPOWERDOWN=10
Recv: // ========== Queried registers ==========
Recv: // GCONF:      ffffffff recalibrate=1 faststandstill=1 en_pwm_mode=1 multistep_filt=1 shaft=1(Reverse) diag0_error=1 diag0_otpw=1 diag0_stall=1 diag1_stall=1 diag1_index=1 diag1_onstate=1 diag1_steps_skipped=1 diag0_int_pushpull=1 diag1_poscomp_pushpull=1 small_hysteresis=1 stop_enable=1 direct_mode=1 test_mode=1
Recv: // CHOPCONF:   ffffffff toff=15 hstrt=7 hend=15 fd3=1 disfdcc=1 chm=1 tbl=3 vhighfs=1 vhighchm=1 tpfd=15 MRES=15(0usteps) intpol=1 dedge=1 diss2g=1 diss2vs=1
Recv: // GSTAT:      ffffffff reset=1(reset) drv_err=1(ErrorShutdown!) uv_cp=1(Undervoltage!)
Recv: // DRV_STATUS: ffffffff SG_RESULT=1023 s2vsa=1 s2vsb=1 stealth=1 fsactive=1 CSACTUAL=255 stallGuard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
Recv: // FACTORY_CONF: ffffffff FACTORY_CONF=31
Recv: // IOIN:       ffffffff REFL_STEP=1 REFR_DIR=1 ENCB_DCEN_CFG4=1 ENCA_DCIN_CFG5=1 DRV_ENN=1 ENC_N_DCO_CFG6=1 SD_MODE=1 SWCOMP_IN=1 VERSION=0xff
Recv: // LOST_STEPS: ffffffff LOST_STEPS=1048575
Recv: // MSCNT:      ffffffff MSCNT=1023
Recv: // MSCURACT:   ffffffff CUR_A=-1 CUR_B=-1
Recv: // OTP_READ:   ffffffff OTP_FCLKTRIM=31 otp_S2_LEVEL=1 otp_BBM=1 otp_TBL=1
Recv: // PWMCONF:    ffffffff PWM_OFS=255 PWM_GRAD=255 pwm_freq=3 pwm_autoscale=1 pwm_autograd=1 freewheel=3 PWM_REG=15 PWM_LIM=15
Recv: // PWM_SCALE:  ffffffff PWM_SCALE_SUM=255 PWM_SCALE_AUTO=-1
Recv: // PWM_AUTO:   ffffffff PWM_OFS_AUTO=255 PWM_GRAD_AUTO=255
Recv: // TSTEP:      ffffffff TSTEP=1048575
```

jalanjarosz, 2019-10-22:
@vladbabii 
If your CoreXY has 3 Z-steppers, then remove the single Z-Stepper from SKR 2, at lease comment out in your config. 
If your CoreXY has 4 Z-Steppers, then remove  the single Z-Stepper from SKR 2 and install and configure on SKR 3. Also add your z-endstop/sensor to SKR 3. 

I have 2 MCU on my CoreXY with all 4 z-Steppers on the same MCU.  You can still add heaters and extruder to both SKR boards.

vladbabii, 2019-10-22:
i have no z stepper on board 2. i said that i use stepper in socket for z stepper. as i said above
- no steppers on board 1
- X and Y steppers and one extra tmc2208 in standalone for testing are on board 2
- Z, z1 and z2 steppers are on board 3

Anyway, i'm running only board 2 now, the following config with only one board, same result (only ffff....f in values).

Do i need some specific pins or should any digital pins work for software spi?

```
[stepper_x]
step_pin: PC6
dir_pin: PC7
enable_pin: !PB15
step_distance: .0025
endstop_pin: PC2 # X+ is PA2
position_endstop: 0
position_max: 200
homing_speed: 50

[tmc5160 stepper_x]
cs_pin: PC15
#spi_bus: spi1
#spi_speed: 2000000
spi_software_sclk_pin: PB7
spi_software_mosi_pin: PA6
spi_software_miso_pin: PA7

microsteps: 16
interpolate: True
run_current: 0.9
stealthchop_threshold: 100

[stepper_y]
step_pin: PB13
dir_pin: PB14
enable_pin: !PB12
step_distance: .0025
endstop_pin: PC1 # Y+ is PA1
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_z]
step_pin: PB10
dir_pin: PB11
enable_pin: !PB2
step_distance: .0125
endstop_pin: PC0 # Z+ is PC3
position_endstop: 0.5
position_max: 200

[extruder]
step_pin: PC5
dir_pin: PB0
enable_pin: !PC4
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250

#[heater_bed]
#heater_pin: PC9
#sensor_type: ATC Semitec 104GT-2
#sensor_pin: PB1
#control: watermark
#min_temp: 0
#max_temp: 130

[fan]
pin: PC8

[mcu]
serial: /dev/printerhill_0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100
```

KevinOConnor, 2019-10-23:
> The 2209 with uart work fine with :pin for uart but the :spi does not seem to work. I tried x:spi and x:spi1.

The mcu is selected by the cs_pin - so use something like `cs_pin: xy:PA123`.  The cs pin must be on the same micro-controller as the spi bus (but does not need to be on the same micro-controller as the stepper dir/enable/step pins).

> Do i need some specific pins or should any digital pins work for software spi?

Any digital output pins should work with software spi.

> Anyway, i'm running only board 2 now, the following config with only one board, same result (only ffff....f in values).

It's hard to say, but sounds like a wiring or incorrect pin configuration.

-Kevin

KevinOConnor, 2019-10-23:
> but does not need to be on the same micro-controller as the stepper dir/enable/step pins

To be clear, using a different mcu to configure the driver would be an odd choice and probably not the best plan.  Definitely don't mix 5V (mega) with 3.3V (skr) wiring as I suspect that could cause damage to the chips.

-Kevin

Hywelmartin, 2019-10-23:
So you are using tmc5160 om the mini 1.1.. there isn't any pins that are connected in the stepper socket for the SPI communication.. #1896

vladbabii, 2019-10-23:
@Hywelmartin i used wires. The stepstick pins are soldered on top of the step stick, not bottom so they don't connected to anything on the board, then i used dupont wires to connect pins on board to the step stick

vladbabii, 2019-10-23:
@KevinOConnor 
The step sticks are these https://learn.watterott.com/silentstepstick/pinconfig/tmc5160/ - version 1.4
The SDI to DCO pins (#2 to #6 on left side) are soldered up on step stick, not connected to board.

[image]

Maybe i misunderstood the wiring, it's my first time using spi

data lines on step stick
- SDO/CFG0 | MISO - Serial Data Output
- SDI/CFG1 | MOSI - Serial Data Input

data lines on board: defined via software_spi_... settings

vladbabii, 2019-10-24:
Maybe the issue is that v 1.4 has no CLOCK pin ? on 1.4 DCO replaced the 1.3 CLK pin

vladbabii, 2019-10-24:
Sorry everyone for wasting your time. The board is broken somewhow. I tried same firmware + settings on another skr mini 1.1  and it works fine. Surprinsingly, uart works fine on the broken board.

Thank you for your help, I'm gonna go burn a board now...

closed 2019-10-24 by vladbabii