Klipper3d/klipper #2100: PI 3b -> SKR 1.3 using GPIO TX/RX pins

MaxDeAngelis, 2019-10-23:
Hello,

First off let me say I love Klipper, so thank you! I apologize if this is the incorrect forum to ask a question like this but I'm really having a hard time finding help here and I saw similar issues raised so figured it was worth a shot.

So I installed Klipper and connected successfully using my USB from PI to SKR 1.3. This works great. I have actually dialed in some of my settings and have it printing pretty well.

The problem is that I would like to connect my PI to my mainboard using the TX/RX GPIO pins. This is how I had it prior to Klipper and I really dont have room in my case for a USB :/ 

Is this something that "should" work?

I feel like I have tried everything I can think of so if you have any pointers that would be great!

Things I have tried:
1. Generate klipper.bin that has just defaults using USB, this works when plugging in a USB. If I point my printer.cfg to "/dev/ttyAMA0" it fails to connect 
2. Generate klipper.bin that is set not to use USB but this fails to install on my board... my beeper just runs forever...
3. Tried any combination of the two above options
4. Updated my printer.cfg to every port I can think of like dev/serial0 and /dev/ttyAMA0 and any other I saw

Also, by the way I have an Ender 3 if that matters. 

Anyway, any help or suggestions would be greatly appreciated \.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3759322/klippy.log.zip)

KevinOConnor, 2019-10-23:
As far as I know, it should work.  @b-vs also previously reported success.  The AUX-1 (or TFT) header on the SKR should be wired to the RPi GPIOs, the `serial` setting in printer.cfg should be set to `/dev/ttyAMA0`, the USB option should be disabled in "make menuconfig", and the firmware should be compiled and flashed to the board.

If you're still having issues after that, attach the klipper log file from that attempt, along with the "~/klipper/.config" file.

-Kevin

MaxDeAngelis, 2019-10-23:
Well I happy it is at least possible 😃 Still no luck. When I boot my machine I need to have the USB attached otherwise it never boots, not sure if that means anything. I assume it's just because it is failing to connect. Once booted I then disconnect the USB and try a firmware restart from OctoPrint to force it to connect through the TX/RX. 

I have attached the files you mentioned.
[printer.cfg.zip](https://github.com/KevinOConnor/klipper/files/3759568/printer.cfg.zip)
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3759569/klippy.log.zip)
[config.txt](https://github.com/KevinOConnor/klipper/files/3759572/config.txt)

KevinOConnor, 2019-10-23:
Your log indicates Klipper successfully contacted the micro-controller and established communications.  Unfortunately, it was unable to issue a micro-controller restart - to fix this, add `restart_method: command` to the mcu section of the config (see config/example.cfg for further details).

> When I boot my machine I need to have the USB attached otherwise it never boots

I'm not sure what you mean, but it certainly sounds odd.  It could be a general indication of wiring issues..

-Kevin

MaxDeAngelis, 2019-10-24:
Holy cow! It's all working now 😃 

I think my comment about needing the USB plugged in was just me being confused by the fact the restart not working. I gave the example.cfg a good read, made the update you suggested, unplugged my USB and restarted and now it boots fine and I can restart the micro-controller.

Thanks so much for the help here!

closed 2019-10-24 by MaxDeAngelis