Klipper3d/klipper #2102: Stepper problems - Cetus/TinyFab CPU

dharvey242, 2019-10-23:
Team,
The config file seems to load okay.  I can get the fan to start/stop, bltouch, even the LCD is okay..ish.  However when I start the device it energizes the stepper motors but if I can to move them it seems to restart the printer (G28).  They won't move under any circumstance.  The only check so far was to ensure power was getting to both the coils in each motor which it is (24v).   For reference i've reset the machine to run on just smoothieware and the steppers etc are fine.  So not such when I switch to the klipper firmware it seems to have an issue.  All help is much appreciated. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3760897/klippy.log)

BlackStump, 2019-10-23:
have you tried removing ! from the stepper_enable pins ?

dharvey242, 2019-10-23:
now i feel stupid....worked.   I was certain i tried that previously....thank you.