Klipper3d/klipper #2103: Adding encoder feedback

spaxxomatic, 2019-10-23:
Hi @Kenzu,
I just discovered Klipper a couple of days ago, while developing the exact same idea: raspi as motion planner, avr as stepping timer. I am currently using an AVR connected via SPI  as a pure freq generator.
 
First of all, big kudos for your outstanding work. High quality code, documentation, smart implementation. 
I'd like to contribute by adding linear scales support. I'm building CNC mills and lathes that need more precision than open-loop steppers can provide, and linear encoder feedback is a must. 

My idea was to have a different thread that computes the delta expected pos - actual position and adds step entries in the queue at the right place. Given the internals of Klipper (iterative solver, look-ahead, etc) this might not be that easy, I suspect. So. before I jump into the code and start tinkering,  maybe you could give me some hints on how you would do it or if there is some work already done in that direction. 
Thx !

KevinOConnor, 2019-10-23:
Thanks.

There's been some interest in closed-loop positional feedback in the past.  Nearly a year ago, I added some initial support for the "mechaduino" - see PR #1038 .  Alas, I haven't had a chance to revisit that since.

The idea with the mechaduino branch was to add in additional steps directly at the micro-controller - which would provide fast response times to encoder feedback.  If, however, the sensor you're working with doesn't require super fast response times (it is okay with 100ms+ response) then implementing in the host is likely simpler.

-Kevin

spaxxomatic, 2019-10-23:
PS: The industrial controls have two loops: the servo position is measured direct at the motor axis via a resolver and used in a fast loop to compensate for sudden load changes, to track the speed and control the commutation, and the linear scales loop are in a second, slower loop, and are used for tracking the real position and adjusting the PWM (or whatever information is sent to control the motor speed). Nicely explained here https://www.motioncontroltips.com/why-is-the-bandwidth-of-a-servo-control-loop-important/

KevinOConnor, 2019-10-25:
Interesting.

FYI, the Klipper host software doesn't run with any special priorities (and it runs on regular Linux distributions).  For sensors routed through the host code, the micro-controller code provides a timestamp on the sensor data, the host figures out the response, and then it schedules the appropriate micro-controller updates at least 100ms from the sensor timestamp.  Thus comms jitter and scheduling jitter don't impact the response.  This obviously increases the response time to sensor data, but the increase is static.

Currently, all robotics input and output to the host code is done via the micro-controller code.  It is possible to run the micro-controller code on a Linux machine (see src/linux/ for the details), but we currently only do that for signals that don't require very precise timing.

-Kevin

closed 2019-12-01 by KevinOConnor