Klipper3d/klipper #2104: lag issue with m104 when printing

phil3412, 2019-10-23:
hi there
i got an issue with both on my printers, anycubic i3 and ender3
while im printing, i got a lag issue issued with m104
each time the printer read m104, the print stop, make a big dot then continue after 2 or 3 seconds
printers running on running on rpi 3b and 3b+
as you can see in the last line of the log... ready_bytes=4125 stalled_bytes=8590
i think those stalled bytes make the lag issue, but where those one come from
thanks for your help
[klippy.log](https://github.com/KevinOConnor/klipper/files/3764771/klippy.log)

tmaniac, 2019-10-24:
Did you just update Octoprint to version 1.3.12?

i'm noticing the same behaviour during my print atm.

everytime the temp dips below my set temp (215c) the hotend stops for a second or so and then continues.

the print is still on-going but i'll check the logfile later on.

phil3412, 2019-10-24:
ohh i will update asap, the update just appear
thanks

closed 2019-10-24 by phil3412

tmaniac, 2019-10-24:
hmm, that's not the solution for me... :(