Klipper3d/klipper #2105: Printer is not ready

RaoulFabisch, 2019-10-24:
I just setup klipper and have the problem that klipper shows me that the printer is not ready:
12:08:45 Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready
12:08:45 Printer is not ready
12:09:00 Reloading Klipper Configuration.
12:09:04 Klipper state: Disconnect

Where could be the error? I checked the port and it seems to be right.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3767196/klippy.log)

FHeilmann, 2019-10-24:
We'll need some info about how you connect your printer controller to your raspberry pi. Klipper tries to opan serial port on `/dev/ttyS0` but doesn't get a response.

RaoulFabisch, 2019-10-24:
solved the issue. Now it works :)

closed 2019-10-24 by RaoulFabisch