Klipper3d/klipper #2108: 40:1 geared remote direct drive microsteps or steps per mm

johngianni, 2019-10-24:
Dear all,
i've just installed the flex3dire remote dd extruder on my cr-10s (board v2.1). I'm now trying to calibrate the extrution steps per mm, but it seems I'm not getting to the correct setting, as lowering too much the step_distance value would cause the motor to go south (like it make a big noise with no move). The Flex3Drive manual says that one should set the microstepping as follows:

> You need to set the esteps to 1900 and set microstepping to x8
If you cannot set microstepping to x8 then run with 3800 esteps if using x16. Do not use any other value for esteps as it will be incorrect

How to do this on the CR-10s with A4988 driver, please?
[klippy.log](https://github.com/KevinOConnor/klipper/files/3769714/klippy.log)

BlackStump, 2019-10-25:
AFAIK the microsteps on that board is fixed at 16 so the step distance is 1/3800

johngianni, 2019-10-25:
Thanks for getting back... uhhhh that's bad news indeed... It seems I'm short of about 20% (1cm on 5 cm extrusion)... U think it would be possible to correct it through flow rate?

nophead, 2019-10-25:
I don't see why the step distance would be fixed as it depends on how far the hobbed drive bites into the filament, so needs to be calibrated per filament type in my experience. I don't see why Flexdrive would be any different in that respect.

nophead, 2019-10-25:
Have you tried reducing the motor current? It is counter intuitive but that is what the designer recommended to me. I haven't actually tried to use mine yet. I might use a servo instead of a stepper motor.

johngianni, 2019-10-25:
> Have you tried reducing the motor current? It is counter intuitive but that is what the designer recommended to me. I haven't actually tried to use mine yet. I might use a servo instead of a stepper motor.

Yep, now at 0,440

johngianni, 2019-10-25:
Unfortunately not... with the current 0,00028 value I'm at the limit. Sometimes the motor stops for a short while during the print (when it shouldn't...)

johngianni, 2019-10-25:
I've managed to go lower with the step_distance... Now @0,000234. This gives perfect extrusion distance. Solution was to fine tune the VREF to a slightly lower value and reduce the e max velocity. Thanks all for the suggestions...

closed 2019-10-25 by johngianni