Klipper3d/klipper #2109: Stepper motor steps/mm

dharvey242, 2019-10-25:
Okay so we're moving on the X,Y and Z axis no problem.  I've even got the BLTouch working fine.  

However

My smoothieware config says 160 steps per mm on all 3 axis.  Which when converted to klipper ( 1/160) gives me 0.00625‬.  If i plug that number in to the Klipper config file then the movement is around half that of what it should be.  i.e. it moves 5mm when it should move 10mm.  But not exactly half.

Also the axis do not move the same distance even though the distance per step is the same.  I'm thinking the current is not correct?  or the step distance is not right in the firmware - should be 32.

How do I set the current? From what I understand I have DRV8824 drivers.

Can I just add 
run_current: 1.5
somewhere in the config?

BlackStump, 2019-10-25:
what board are we talking about? could you adjust the current in smoothieware?
step distance is exactly that. The distance 1 step moves.
>or the step distance is not right in the firmware - should be 32.
I presume you are talking about microsteps? is that smoothieware controlled or hardware controlled?
It would be helpful if you included your smoothieware config and your klipper log and some further details about your hardware.

dharvey242, 2019-10-25:
This is the appropriate parts of the smoothieware config

# Basic motion configuration
default_feed_rate                            4000             # Default speed (mm/minute) for G1/G2/G3 moves
default_seek_rate                            4000             # Default speed (mm/minute) for G0 moves
mm_per_arc_segment                           0.0              # Fixed length for line segments that divide arcs, 0 to disable
#mm_per_line_segment                         5                # Cut lines into segments this size
mm_max_arc_error                             0.01             # The maximum error for line segments that divide arcs 0 to disable
                                                              # note it is invalid for both the above be 0
                                                              # if both are used, will use largest segment length based on radius

# Arm solution configuration : Cartesian robot. Translates mm positions into stepper positions
alpha_steps_per_mm                           160               # Steps per mm for alpha stepper
beta_steps_per_mm                            160               # Steps per mm for beta stepper
gamma_steps_per_mm                           160               # Steps per mm for gamma stepper

# Planner module configuration : Look-ahead and acceleration configuration
# See http://smoothieware.org/motion-control
acceleration                                 3000             # Acceleration in mm/second/second.
#z_acceleration                              500              # Acceleration for Z only moves in mm/s^2, 0 uses acceleration which is the default. DO NOT SET ON A DELTA
junction_deviation                           0.05             # See http://smoothieware.org/motion-control#junction-deviation
#z_junction_deviation                        0.0              # For Z only moves, -1 uses junction_deviation, zero disables junction_deviation on z moves DO NOT SET ON A DELTA

# Cartesian axis speed limits
x_axis_max_speed                             30000            # Maximum speed in mm/min
y_axis_max_speed                             30000            # Maximum speed in mm/min
z_axis_max_speed                             300              # Maximum speed in mm/min

# Stepper module configuration 
# Pins are defined as  ports, and pin numbers, appending "!" to the number will invert a pin
# See http://smoothieware.org/pin-configuration and http://smoothieware.org/pinout
alpha_step_pin 2.1 #                         2.0              # Pin for alpha stepper step signal
alpha_dir_pin 0.11 #                         0.5              # Pin for alpha stepper direction
alpha_en_pin                                 2.4!             #0.4 # Pin for alpha enable pin
alpha_current                                1.5              # X stepper motor current
alpha_max_rate                               30000.0          # mm/min

beta_step_pin 2.0 #                          2.1              # Pin for beta stepper step signal
beta_dir_pin 0.5! #                          0.11             # Pin for beta stepper direction
beta_en_pin                                  2.4!             #0.10  # Pin for beta enable
beta_current                                 1.5              # Y stepper motor current
beta_max_rate                                30000.0          # mm/min

gamma_step_pin                               2.2              # Pin for gamma stepper step signal
gamma_dir_pin 0.20! #                        0.20!            # Pin for gamma stepper direction
gamma_en_pin                                 2.4!             # 0.19 # Pin for gamma enable
gamma_current                                1.5              # Z stepper motor current
gamma_max_rate                               300.0            # mm/min

dharvey242, 2019-10-25:
Sorry its a Cetus running a TinyFab CPU.

dharvey242, 2019-10-25:
drivers are DRV8824.

BlackStump, 2019-10-25:
I think you would need to know how the stepper current is controlled on your hardware to then know if that is supported in klipper, a schematic of your hardware would be helpful.

dharvey242, 2019-10-25:
I don't have a complete board schematic only the cpu.  Would that be of use? It uses electronic control rather than physical pots and i've read somewhere that this board in particular doesn't use i2c. 
[TFUPCPUV5R_1(1).pdf](https://github.com/KevinOConnor/klipper/files/3771965/TFUPCPUV5R_1.1.pdf)

dharvey242, 2019-10-25:
sorry the guy that created the cpu has just told me the current is set by the hardware so the in the smoothieware config it does nothing.

BlackStump, 2019-10-26:
It would seem you need to physically look at your board and locate the stepper drivers and then find the pot.
I know nothing of your hardware and it seems there is little to no info on the board in the wild, you are at the mercy of the manufacture to disclose details of there closed sourced products.
someone with more knowledge on your hardware may chime in.

vladbabii, 2019-10-27:
Here is how you set current on drv8825

https://www.youtube.com/watch?v=YJSmrbeZUp4

For setting correct step values you can use this simple form below, just tell it to move a distance, measure the actual distance moved and plug those values in the form. It will spit out the correct settings. For drv8825 the steps values are influenced by jumpers ms1/ms2/ms3

https://jsfiddle.net/wladq66/ejyh489t/21/

dharvey242, 2019-10-27:
Thanks for the input, it's appears the current is correct as it's set by the hardware and that element hasn't been changed so I can only assume my unusual steps issues must be software related.  The XY and z axis all have the same stepper, driver motor and motion system. So each axis if they have the same distance per step applied I would have assumed they would move the same distance but they dont.  Bit stuck now.

vladbabii, 2019-10-27:
Just measure the move size and set the values accordingly.  You won't lose anything.

dharvey242, 2019-10-27:
Yeah I started doing that but wanted a more mathematical answer.  Feels odd they might be different when previously under the smoothie config they were all exactly the same.

vladbabii, 2019-10-27:
Are you sure it did not toggle microstep pins ? that would explain it

dharvey242, 2019-10-27:
Sorry not sure what you mean.  Some of this is beyond my knowledge..

dharvey242, 2019-10-28:
This post seems to have explain the exactly the same issues i'm having:
https://reprap.org/forum/read.php?13,303570

The key part of the solution in the post was:
"Thanks, bobc, for that suggestion! The configuration.h file for the Repetier firmware has a define STEPPER_HIGH_DELAY, which I changed from zero to one to make the step pulses sent to the drivers last microsecond longer. That fixed my problem! Now all my axes move exactly as far as they're supposed to, and the motors sound much happier when they move too. Cripes, I don't know how long it would have taken me on my own to think to look at the signal timing."

Is this configuration to be found in Klipper somewhere?

vladbabii, 2019-10-28:
I don't know why repetier needed that settings, neither klipper nor marlin need something like that to work with 8825 drivers.

dharvey242, 2019-10-28:
It has 8824 drivers which are supposed to be near identical to the 8825 but that post suggests otherwise. I'm at a loss now as to how to proceed.

vladbabii, 2019-10-28:
You just need to calculate the new values and put them in config. I would not give it too much attention unless you have an issue like skipped steps, steppers not moving or making weird noises.

dharvey242, 2019-10-29:
Okay I calculated the steps but its apparent when I do try and print that in particular the x-axis is missing steps. The finished cube is all over the place on the x-axis.  And you can hear physically a grinding sound on the x as if its starting and stopping.  The other axis seem to be fine.  Very odd given each axis has the same driver, stepper and movement.

KevinOConnor, 2019-10-29:
That's really odd.  The DRV8824 spec says it needs 1.9us and we haven't seen any complaints on the common DRV8825.  Maybe the DRV8824 is less tolerant than the spec states.

In any case, there isn't any harm in increasing the step delay, so if that makes it work then great.

-Kevin

closed 2019-11-01 by klipper-gitissuebot