Klipper3d/klipper #2110: Bed probe offsets are not being calculated to the output mesh causing large deviations?

JohnEdwa, 2019-10-25:
And no, this is not the "it's probing with the nozzle" issue.

Because of the offset with my BLtouch probe (X -46.5, Y -9) and the restrictions on the frame on my printer movement I cannot probe the whole 235x235 bed edge-to-edge - I can move the nozzle between X0 and X240 so I ended up with ``max_point: 240,202.5`` and ``min_point: 88, 50.5`` to get a roughly symmetrical area, meaning there is ~45mm margin around the edges.

However, after I run a probe and query for the bed mesh, I receive the output shown below with the measured corner/outside values being directly placed into the outside of the mesh.

So even though the actual probing positions were all 45mm inside the build plate dimensions, the mesh is stretched to fill the whole 235x235 bed area and the result that is supposed to be at X45 Y45 is now at X0 Y0 and so on. This doesn't affect much on small offset or low probing counts, but with 6 probes on a 235mm bed with 45mm margins, the *actual* distance between the probes is just 24mm, twice the error caused here.

Is this just a bug on the BED_MESH_OUTPUT function, or does Klipper do this internally as well? 

```
Recv: // Mesh Leveling Probed Z positions:
Recv: // -0.021250 -0.142500 -0.210000 -0.172500 -0.212500 -0.052500
Recv: // -0.181250 -0.052500 -0.205000 -0.187500 -0.195000 -0.190000
Recv: // -0.162500 -0.166250 -0.056250 -0.181250 -0.213750 -0.217500
Recv: // -0.167500 -0.165000 -0.198750 -0.070000 -0.237500 -0.207500
Recv: // -0.167500 -0.143750 -0.211250 -0.200000 -0.107500 -0.202500
Recv: // -0.056250 -0.127500 -0.162500 -0.185000 -0.212500 -0.045000
Recv: Mesh X,Y: 16,16
Recv: Search Height: 7
Recv: Mesh Average: -0.16
Recv: Mesh Range: min=-0.1977 max=0.3202
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   -0.056250  -0.079376  -0.105074  -0.127500  -0.144211  -0.155295  -0.162500  -0.168368  -0.175362  -0.185000  -0.196984  -0.208330  -0.212500  -0.198532  -0.150171  -0.045000
Recv:   -0.110802  -0.245428  -0.221098  -0.148297  -0.092506  -0.082812  -0.120520  -0.187762  -0.256109  -0.295180  -0.281255  -0.205882  -0.084491  0.034999  0.063565  -0.137527
Recv:   -0.147305  -0.221417  -0.196624  -0.147850  -0.117513  -0.122051  -0.158441  -0.210727  -0.256537  -0.273611  -0.246323  -0.172202  -0.068457  0.021500  0.021532  -0.183748
Recv:   -0.167500  -0.135189  -0.130300  -0.143750  -0.166814  -0.191555  -0.211250  -0.220816  -0.217239  -0.200000  -0.171500  -0.137491  -0.107500  -0.095255  -0.119117  -0.202500
Recv:   -0.174707  -0.063621  -0.079964  -0.144763  -0.207852  -0.242847  -0.242140  -0.211871  -0.166917  -0.125874  -0.106041  -0.118397  -0.162593  -0.221926  -0.258330  -0.207353
Recv:   -0.173184  -0.042222  -0.070572  -0.152819  -0.224432  -0.254254  -0.236975  -0.185624  -0.124041  -0.079367  -0.074524  -0.120692  -0.209799  -0.306999  -0.343151  -0.207308
Recv:   -0.167500  -0.074726  -0.102131  -0.165000  -0.213234  -0.224729  -0.198750  -0.149304  -0.098515  -0.070000  -0.082241  -0.141962  -0.237500  -0.332184  -0.357707  -0.207500
Recv:   -0.161898  -0.142692  -0.157626  -0.175535  -0.180307  -0.167681  -0.142047  -0.113240  -0.093340  -0.093467  -0.120584  -0.174288  -0.243611  -0.303819  -0.313206  -0.209894
Recv:   -0.159666  -0.215103  -0.210951  -0.177788  -0.137581  -0.104797  -0.087500  -0.088464  -0.106271  -0.136420  -0.172426  -0.206933  -0.232809  -0.244258  -0.237922  -0.213985
Recv:   -0.162500  -0.257961  -0.234842  -0.166250  -0.099371  -0.059863  -0.056250  -0.084314  -0.131488  -0.181250  -0.217514  -0.229024  -0.213750  -0.183275  -0.167193  -0.217500
Recv:   -0.169877  -0.243889  -0.208804  -0.138529  -0.078876  -0.052596  -0.064408  -0.106039  -0.161251  -0.210874  -0.237847  -0.232240  -0.196296  -0.149459  -0.133409  -0.217095
Recv:   -0.178416  -0.161723  -0.127041  -0.097341  -0.084692  -0.092463  -0.117515  -0.152403  -0.187563  -0.213519  -0.223069  -0.213492  -0.188735  -0.161613  -0.156006  -0.209057
Recv:   -0.181250  -0.026115  -0.006391  -0.052500  -0.117308  -0.172507  -0.205000  -0.213285  -0.203839  -0.187500  -0.175852  -0.177610  -0.195000  -0.220147  -0.231457  -0.190000
Recv:   -0.167392  0.112872  0.105751  -0.022912  -0.165618  -0.263174  -0.294640  -0.267290  -0.206561  -0.146013  -0.117280  -0.140026  -0.211898  -0.298484  -0.323265  -0.157569
Recv:   -0.121101  0.160164  0.123495  -0.038560  -0.203421  -0.306133  -0.327022  -0.279343  -0.196944  -0.121914  -0.092239  -0.129461  -0.226327  -0.334448  -0.351951  -0.111135
Recv:   -0.021250  -0.033920  -0.085468  -0.142500  -0.185926  -0.208107  -0.210000  -0.198302  -0.182598  -0.172500  -0.174799  -0.190607  -0.212500  -0.221667  -0.185051  -0.052500
```

JohnEdwa, 2019-10-25:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3773071/klippy.log)

Arksine, 2019-10-26:
The mesh is not stretched to fill the whole bed area, in truth it has no way to do so as it doesn't know the where the bed starts and ends.

I'm not sure I see the problem with the output.  It doesn't label the x,y coordinates of the points.  For the mesh output, the y-axis is inverted.  This was done to keep output compatibility with other firmware.  Its possible that this is confusing, as you might expect the top-right value for the probe and mesh output to be the same.  If you look closer, the top-right value for the probed output matches the bottom-right value of the mesh, and so on.

JohnEdwa, 2019-10-26:
Here's the output with ``min_point: 128, 64`` and ``max_point: 200,180`` with a 3x3 grid, so this is probing an area inside the bed that is just roughly 100x100mm. I placed a ruler at the back middle point, so it is way off as you can see:

> 
> Send: BED_MESH_OUTPUT
> Recv: // Mesh Leveling Probed Z positions:
> Recv: // 0.002500 -0.002500 -0.065000
> Recv: // -0.002500 -0.047500 -0.053750
> Recv: // -0.022500 **0.682500** -0.068750
> Recv: Mesh X,Y: 7,7
> Recv: Search Height: 5
> Recv: Mesh Average: 0.05
> Recv: Mesh Range: min=-0.1686 max=0.6325
> Recv: Interpolation Algorithm: lagrange
> Recv: Measured points:
> Recv:   -0.022500  0.374306  0.609306  **0.682500**  0.593889  0.343472  -0.068750
> Recv:   -0.014167  0.195031  0.317438  0.353056  0.301883  0.163920  -0.060833
> Recv:   -0.007500  0.062994  0.102068  0.109722  0.085957  0.030772  -0.055833
> Recv:   -0.002500  -0.021806  -0.036806  -0.047500  -0.053889  -0.055972  -0.053750
> Recv:   0.000833  -0.059367  -0.099182  -0.118611  -0.117654  -0.096312  -0.054583
> Recv:   0.002500  -0.049691  -0.085062  -0.103611  -0.105340  -0.090247  -0.058333
> Recv:   0.002500  0.007222  0.005556  -0.002500  -0.016944  -0.037778  -0.065000

That exact measurement is at the edge of the interpolated grid, so it produces this visualization, which  is *not* correct, as the 0.682500mm value is not supposed to be at x118 y0, it's at x118 y**64**.

[image]

So I ask again, is this a bug in the BED_MESH_OUTPUT, and Klipper is internally iterpolating the bed from Y64 down to Y0 and placing that correction at the proper position at Y64, or is it taking that 100x100 mesh, stretching it to fill the whole 235x235 bed, and that correction is now wrongly at the edge of the bed?

Arksine, 2019-10-26:
Once again, Klipper does not stretch the mesh internally.  It uses the probe offsets to map the probed area to toolhead coordinates.  All areas outside of the mesh are clamped to the nearest edge of the mesh.

Unless Marlin has changed G29 output recently it behaves the same way.  Its output does not stretch over the entire bed.  Marlin has an option to interpolate past the edges of the mesh (which IMO is dangerous), but that interpolation is not represented in the mesh output.

As for the visualization, first of all its inverted, so you need to configure the visualizer to flip the Y axis.  The value for .682500 should be at Y180.  Its real coordinate is at Y171 to account for the probe offset.  Of course the visualizer knows none of this, all it knows are the number of Z values it receives on each axis, and the bed size retrieved from Octoprint's configuration.  The visualizer stretches the mesh over the entire bed, however you can think of it as a visualization of the portion of the bed actually probed.

JohnEdwa, 2019-10-27:
Ah, right.   
So it is just the visualization that is wrong because the Klipper command doesn't tell the size of the probed area, but it also doesn't actually make a mesh that would fit the whole bed.

Guess I kinda assumed that [as example output on the Bed Visualizer page has the data corner coordinates](https://plugins.octoprint.org/plugins/bedlevelvisualizer/), if there is no size data with the mesh output then it obviously should be the full size of the bed to make any sense. It didn't really pop into my mind that not making sense was an option.

And if it also doesn't do any interpolation outside the probed area whatsoever and instead just repeats the closesest the value as is, then symmetry doesn't really matter either and I should probe as close to all the edges I can. Feels pretty odd though as that means it never actually probes the exact middle of the bed.

KevinOConnor, 2019-11-06:
Okay, I'm going to close this as it sounds like the original question was answered.

-Kevin

closed 2019-11-06 by KevinOConnor