Klipper3d/klipper #2112: MCU 'mcu' shutdown: No next step

effernity, 2019-10-26:
Hello

I was printing simple object about 40 mm height and when reached 30 mm there was an error as described in title, log attached. Also, I'm getting "timer too close" error quite often when print multiple objects one after another, on 3rd or 4th print there's a big chance that after start i will immediately get "timer too close"
[klippy.log](https://github.com/KevinOConnor/klipper/files/3774230/klippy.log)

andreklug, 2019-10-26:
Are you printing with high speeds?

effernity, 2019-10-26:
This particular print was at 50 mm/s

effernity, 2019-10-26:
Just started print, after few layers - timer too close
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3775120/klippy.zip)

KevinOConnor, 2019-10-26:
It looks like you've done a custom install of the host software.  The first log indicates a shutdown that resulted because the software didn't get any processing time for over 5 seconds.  The host software does have some real-time requirements - if the software is installed in a container, on a machine with other high priority processes that may be using excessive cpu, or on a machine that runs into swap, then the software is unlikely to function correctly.

-Kevin

effernity, 2019-10-26:
> It looks like you've done a custom install of the host software. The first log indicates a shutdown that resulted because the software didn't get any processing time for over 5 seconds. The host [...]

Hello
What does it mean "custom install"? My octoprint is installed by the official installation in venv, klipper just installed by guide on github

KevinOConnor, 2019-10-26:
It's hard to tell from the info in the logs, but it didn't look like the software was running on a typical Raspberry Pi.  What is the host operating system and host machine type?

-Kevin

effernity, 2019-10-26:
It's Orange Pi Lite running Armbian Buster

KevinOConnor, 2019-10-26:
I don't have any first hand experience with that particular board.  It's certainly enough to run Klipper, but it may not be powerful enough to run Klipper and other processes.  Make sure that no other regular process is running at a higher priority than Klipper.  (Some of the OctoPrint installers will give OctoPrint a high priority - this is wrong and will cause problems - if that is the case for your install, then either raise Klipper to the same or higher priority, or make sure OctoPrint runs at normal priority.)  Similarly, make sure no program or process on the machine is causing the machine to go into swap.

-Kevin

effernity, 2019-10-26:
> I don't have any first hand experience with that particular board. It's certainly enough to run Klipper, but it may not be powerful enough to run Klipper and other processes. Make sure that no other [...]

According to htop, all octoprint processes runs at -2 nice and 18 priority, klippy.py runs at 20 prioroty and 0 nice. How to change the default priority?

KevinOConnor, 2019-10-26:
The "-2" nice for OctoPrint can cause problems.  (It's unclear if it is the sole cause of problems, however.)  Alas, I don't have any help for altering priorities - you (or someone else) would need to dig through the OctoPrint and/or Klipper startup scripts to alter them.

-Kevin

effernity, 2019-10-27:
> The "-2" nice for OctoPrint can cause problems. (It's unclear if it is the sole cause of problems, however.) Alas, I don't have any help for altering priorities - you (or someone else) would need to dig through the OctoPrint and/or Klipper startup scripts to alter them.
> 
> -Kevin

Ok, thank you! If it won't help, i'll open other issue.

closed 2019-10-27 by effernity