Klipper3d/klipper #2113: [Pause/Resume][Runout Sensor] Runout event detected while there is still filament

FromtonRouge, 2019-10-26:
I did an 8-10 hours print last night and saw a very strange bug with the filament runout detection.

I had 8-9 runout events detected during the print although the filament was still in the runout sensor like the image below.

[image]

I even had a runout event detected just after resuming the print.
How is it possible,  is there something wrong in my configuration file? Could it be an electrical issue on by board (MKS Gen L) or my sensor? Any idea on this?

Thanks

[klippy.log](https://github.com/KevinOConnor/klipper/files/3774735/klippy.log)

Arksine, 2019-10-26:
Most likely its the switch or wiring.  I notice that you have the switch pin inverted, which means that you are likely using a switch that is normally open.  I would recommend using a normally closed switch.  With a NC switch a wiring or mechanical issue won't trigger a runout, it would trigger an Insert which can only happen if the printer is ready/idle.

tonn0297, 2019-10-26:
i have +1000h on the klipper runout detection. i would say its a hardware bug not a software.

FromtonRouge, 2019-10-27:
Ok I desoldered and changed the wiring of the switch and it's now a normaly closed one. Didn't know about this trick, it's simple and it should work. Thanks!

closed 2019-10-27 by FromtonRouge