Klipper3d/klipper #2114: No fast, smooth moves (look-ahead) without extruder?

Spacemarine2018, 2019-10-26:
I want to build a high-speed CoreXY pen-plotter for an art project, which means I only have X and Y, no extruder. Since I want to go up in speed as far as possible, I need look-ahead for the XY-moves. However the doc says: "Klipper implements look-ahead between moves that have similar extruder flow rates. Other moves are relatively rare and implementing look-ahead between them is unnecessary."

Does this mean that there is no look-ahead without extruder movement? Or does "no-movement" also count as "similar extruder flow rate"?

KevinOConnor, 2019-10-26:
The look-ahead processing is done between moves with no extrusion.  (Two moves with no extrusion have "similar (zero) extrusion rates".)

-Kevin

Spacemarine2018, 2019-10-26:
Thanks, that makes it clear!

Can you give any advice on how to make the fastest and smoothest moves? Every sharp corner slows down the movements and introduces vibration, so splitting up a curve into a huge amount of small segments should be ideal when using look-ahead. But what's the limit? When are the segments becoming too small and the look-ahead buffer isn't large enough anymore?

KevinOConnor, 2019-10-26:
There is no limit on the look-ahead buffer - it will grow to whatever size is needed to accomplish its goals.

With crazy small segmentation, you might get to the point of overloading the Klipper host cpu (or overloading the software feeding the g-code to Klipper).  I don't know where that point is.

I'd guess start with a segment size of 0.100mm - I doubt you'd notice any visual impact at that small size.

-Kevin

Spacemarine2018, 2019-10-27:
Ok, that's great!

How does the smoothed look-ahead come into play? Will the smoothing be triggered earlier or later with segmentation size? Is there any way to influence the threshold when the smoothing will be activated? I want maximum control and predictability of the trajectory, can it be disabled at all?

KevinOConnor, 2019-10-30:
It's possible to disable the accel_to_decel mechanism by setting it to a very large value in the config.

As for the other questions - you'd likely need to build the device and then let us know your findings.

-Kevin

closed 2019-11-15 by KevinOConnor