Klipper3d/klipper #2115: Heavy blobbing/stringing on volcano + petg

antst, 2019-10-27:
My setup: direct drive, geared extruded, vocano, petg, 0.6 nozzle.
Pressure advance is tuned. I also tried with pressure advance smoothing PR, it works in other aspects, but this one it doesn’t change. All “stops” at the end of lines result in quite heavy blobbing and stringing, as those pauses are enough to get some leaking. This is volcano, so we have quite some amount of melted plastic and gravity...pauses aren’t very compatible with volcano. And retract or pressure advance can do little: once we stop, It leaks. With volcano you need to always move. At least with petg.

On the same setup I have nearly zero stringing and totally zero blobbing with marlin (also using pressure/linear advance).

I can try to reduce “stops” by lifting retraction speed and acceleration, but this isn’t good way. Especially with PA.  moreover, with geared extruder acceleration limit is lower, plus pancake stepper will start skipping very quickly.

And, of course, I can try to impose ridiculously high (for direct drive) retract distances, like 6-8 mm, but this will not work and will results in nozzle clogging after some time, I am pretty sure.

And I can’t even imagine how bad it will be on more “typical” volcano nozzles like 0.8 and 1.0.
Those “stops” definitely hurt.

antst, 2019-10-27:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3775670/klippy.log)

antst, 2019-10-27:
It is not that visible if object is simple, but once you try something complicated - it is hell )

BlackStump, 2019-10-27:
are you using Octoprint? and if so are you using the virtual sdcard option?

antst, 2019-10-27:
I am using DWC2 and virtual sdcard.

theophile, 2019-10-27:
I use PETG primarily and I had to experiment a lot to get good results. I'm using Prusa Slicer and I found it helped immensely to enable "avoid crossing perimeters" and "wipe while retracting," and to disable "lift Z."

antst, 2019-10-27:
I never enable "lift Z" with PA :) And always "avoid crossing perimeters". But my observation was made when I was printing something like 6 pieces at once. Then crossing is inevitable :)

And, generally, I didn't have issue with PETG oozing for long time (this is my primary material also). And I was surprised that Klipper behave that bad with Volcano (with normal V6 it is OK).

But it is difficult to not ooze in case of those "stops". Wide nozzle + gravity + large volume of melted plastic. And, of course, those "stops" in klipper are kind of granted when nozzle go to another part, in case you print more than one. It is clearly change of extrusion rate :)

I will try wiping, but I generally dislike it. And with PA  wiping must be obsolete.

nophead, 2019-10-27:
Oozing is greatly dependent on how dry the filament is. I have started keeping the spools that are feeding my printer in in tiny fan ovens that keep them at 50C and 18% relative humidity. It totally stops the nozzle ooze at the end of a print and make bridges a lot tighter. I used to think some ooze was inevitable due to gravity but I think it is actually propelled by steam.

antst, 2019-10-27:
Yep. True. But in this particular case, it is not filament.
1) I tried with few.
2) I tried in current state of everything with Marlin and there is no oozing.

I kind of forgot long time ago about oozing, to be honest ) But this thursday I made yet another attempt to switch to klipper, and found that with Volcano it isn't doing well.
At this moment, I am considering switching back to Marlin again. It is like 4th time when I am trying to stay with Klipper. Generally, I find klipper more interesting than Marlin in many aspects. And in combination with DWC2, I definitely like it more than Octoprint+Marlin (despite loss of all those infinite octoprint plugins). When klipper works, I like it more than Marlin. But I need PETG+Volcano.

KevinOConnor, 2019-10-28:
> I am using DWC2 and virtual sdcard.

This issue tracker is for code found in this repo.  You'll need to reproduce on the pristine code from this repo and provide that log in order for us to assist.

-Kevin

KevinOConnor, 2019-10-28:
> Oozing is greatly dependent on how dry the filament is. I have started keeping the spools that are feeding my printer in in tiny fan ovens that keep them at 50C and 18% relative humidity. It totally [...]

Interesting, thanks.  Separately (and a bit off-topic), I print directly from a ziplock bag with desiccant in it: https://github.com/KevinOConnor/zipspool

-Kevin

antst, 2019-10-28:
> > I am using DWC2 and virtual sdcard.
> 
> This issue tracker is for code found in this repo. You'll need to reproduce on the pristine code from this repo and provide that log in order for us to assist.
> 
> -Kevin

Do you mean to exclude DWC2? I thought that printing from virtual sdcard is fully enclosed within klipper, not?

jakep82, 2019-10-28:
Running DWC2 requires modifications to the gcode.py file, so it is not pristine Klipper code.

KevinOConnor, 2019-10-28:
> Do you mean to exclude DWC2? 

Yes.  There's a lot of interest in a web frontend to Klipper, and I'm also keen to see support added to Klipper.  I know of several developers interested in it as well.  I've had several conversations about it (including with Stefan).  One challenge is getting the threading correct.  As of my last review, dwc2 had threading errors which could cause a variety of theoretical and practical problems.

Beyond that, it is simply not practical to provide support for 3rd party code.  I don't know if the problem will persist on the pristine code from this repo, but determining that is a necessary first step.  Additional things to look at are oddities in the g-code - commands that may cause a flush of the look-ahead buffer - for example: constant changing of fan speed, constant changing of heater temp, frequent changes in extrusion rate, etc.

-Kevin

antst, 2019-10-29:
Oh! Thanks for information! I will retry it again without DWC2 then. Then printing from virtual sdcard must be enough to exclude any external influences, yep?

KevinOConnor, 2019-11-15:
I'm going to close this issue as it looks like the conversation has concluded.

-Kevin

closed 2019-11-15 by KevinOConnor

antst, 2019-12-29:
BTW, this issue was resolved with change in look-ahead.