Klipper3d/klipper #2116: MCU 'mcu' shutdown: Move queue empty

error414, 2019-10-27:
Hello,
I am new with klipper and I try print benchy ship for test klipper. Unfurtunatelly evrytime durring first line I get this error. I use samsung 2A charger for Ri and I replaced USB cable to printer. No success :(.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3776050/klippy.log)

error414, 2019-10-27:
For sure I here is gcode. It is petg and print is pretty slow. 
[3DBenchy.gcode.log](https://github.com/KevinOConnor/klipper/files/3776058/3DBenchy.gcode.log)

error414, 2019-10-27:
For more info I created video for better explanation. https://youtu.be/hHrB5Nz7CbE?t=33
Short while before error I saw "average load: 1.2". I dont know whether it is a problem. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3776129/klippy.log)

error414, 2019-10-30:
Hi,

I tried use virtual SD card, it is better, it means without freezing. But still error during first line. 

Here is capturing during printing 
https://youtu.be/g22-huybpew?t=124

[DBenchy.gcode.txt](https://github.com/KevinOConnor/klipper/files/3790885/DBenchy.gcode.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3790891/klippy.log)

Exactly same gcode I printed with Marlin without any problem.

error414, 2019-11-02:
I tried reinstall octoprint, I use image from octoprint page, and then only install klipper from master branch. No plugins in octoprint only klipper. And problem is there :(. I tried TMC2208 in legacy mode, without success, I have to do something wrong, I am not able to to find out where is problem.

KevinOConnor, 2019-11-05:
It will take me a few days before I'll be able to look at this.

One thing I did notice is that it looks like you have coasting enabled - that causes issues with Klipper - see: https://www.klipper3d.org/Slicers.html

-Kevin

error414, 2019-11-05:
First of all I would like to thank you, for spending your time for this. 

I tried turn of "coast at end" in simplify3d, 
[image]
[3DBenchy.gcode.txt](https://github.com/KevinOConnor/klipper/files/3810565/3DBenchy.gcode.txt) , It looked prommising, it was able to print part of second layer, all errors before happend in first layer or durring changing layer from first to second. 

I tried use prusa slicer too, only basic settings so I dont know whether there is set "coasting", I am not familiar with slic3r.

error414, 2019-11-15:
I bought RPI4 4GB version, original power supply, and new USB cable. Issue is still there. I ordered SKR 1.3 32 bit board, so we will see,

error414, 2019-11-26:
Hi Kevin,

SKR just arrived and I did fast test and it seems new board solved all issues with klipper. 

Petr

closed 2019-11-26 by error414