Klipper3d/klipper #2118: Restart host and firmware restart failures with multiple boards

vladbabii, 2019-10-27:
boards:
* mcu - arduino mega (no ramps, just a board used for "fake" pins - as in nothing attached to them)
* xy - skr mini 1.1
* z - skr mini 1.1

Action: Restarting host or firmware
Expected: klipper/boards to restart and reconnect
Actual result:
* <board> error during connect
*  Failed automated reset of <board> 'xy

Attached:
* log file
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3776205/klippy.zip)
* configuration split into multiple files since it's pretty huge in a single file
[klipper_config.zip](https://github.com/KevinOConnor/klipper/files/3776206/klipper_config.zip)

Log file for these steps:
1. power off boards
2. stop klipper service
3. wait for 5-10 seconds
4. start service
5. connect octoprint
6. everything works fine
7. restart host
8. result > one of the xy or z boards or both cannot be connected to
9. restart firmware
10. result > failed automated reset of MCU xy

Now i do steps 1-7 then quickly turn off then on the power to the mini boards and it works most of the time.

Other details:

* The board are connected to a usb hub that is connected to a laptop.

* OS:
```
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.3 LTS
Release:        18.04
Codename:       bionic
```

Observations / things i've tried
* if i only use the arduino mega board, the firmware reset and host restart works fine, with very rare firmware_reset needed
* using direct connections to laptop instead of a hub makes no difference
* using a pi3 instead of the laptop makes no difference

Thank you

FHeilmann, 2019-10-28:
see if adding 
`restart_method: command` 
to your non-arduino MCUs helps.

vladbabii, 2019-10-28:
Doesn't change anything.

vladbabii, 2019-10-28:
Yes, seems like the same issue.

albmargar30, 2019-10-29:
@vladbabii, Do you have the skr mini v1.1 with TMC2208? The manufacturer does not tell how to connect the TMC in UART mode.

vladbabii, 2019-10-29:
I have with 2209 which are basically the same. I use one pin per stepper driver on the bottom expansion port - i think PC11 and another 2 pins next to it. these are connected to uart pin on stepper driver

ambethia, 2019-11-03:
I have two SKR 1.3s and 2209s. Works on fresh boot, but after trying to issue a `FIRMWARE_RESTART`, I get this in the log:

```
Timeout on serial connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 92, in connect
    identify_data = self._get_identify_data(connect_time + 5.)
  File "/home/pi/klipper/klippy/serialhdl.py", line 57, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 156, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 225, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'identify_response' response
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 138, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 643, in _connect
    raise error(str(e))
error: Unable to connect
```

timmit99, 2019-11-06:
@KevinOConnor  i was having similar issues with not getting my 2 SKR 1.3's to connect and @FHeilmann directed me to try the "work-multimcu-20191104" branch and so far i have not had ANY issues with it. About a dozen or two `FIRMWARE_RESTART`'s without any issues connecting.

vladbabii, 2019-11-06:
Link to pull request for the branch - https://github.com/KevinOConnor/klipper/pull/2144

KevinOConnor, 2019-11-06:
Okay, thanks.  I committed PR #2144.

-Kevin

closed 2019-11-06 by vladbabii