Klipper3d/klipper #2119: Unit for speed? (F100 = 100 mm/s?)

Spacemarine2018, 2019-10-27:
I did not find any information on the unit of speed in the G1 command. Is it mm/s? The documentation doesn't say anything about it

If I issue G1 X100 F100, it moves very slowly. If I set max_velocity in printer.cfg to 100, I can still get a faster movement if I issue G1 X100 F200, so the units in the printer.cfg and the G-code seem to be different. Where can I get more information?

ShohninDmitriy, 2019-10-27:
F100 =100/60=1.66mm/c

Hywelmartin, 2019-10-27:
This is nothing Klipper specific

https://reprap.org/wiki/G-code#G0_.26_G1:_Move

Spacemarine2018, 2019-10-27:
Ok, I see. So the reason is backwards compatibility with slicer software?

In that case it might be a good idea to mention the units in the docs, otherwise it is really counterintuitive of the units in the config file and the g-code commands are different.

tonn0297, 2019-10-28:
feed is a standardized measurement. no need to mention that.

Spacemarine2018, 2019-10-28:
In that case it should be stated which standard is used.

tonn0297, 2019-10-28:
is is the only standard??? feed is derived from mm/s?.... like inch and feet

nophead, 2019-10-28:
It is standard to have feed rate in mm per minute or inches per minute in G-code. It comes from the CNC industry where they talk in inches per minute.

Spacemarine2018, 2019-10-28:
> 
> 
> It comes from the CNC industry where they talk in inches per minute.

It comes from a standard where they talk inches per minute and yet the unit is mm per minute? I couldn't have made a better argument why the units should be explicitly stated. :)

nophead, 2019-10-28:
The measurement units in G code can be switched between inches and mm with a G code. That is why it can be either.

jakep82, 2019-10-28:
Every 3D printer firmware uses mm/min, and the documentation for Klipper provides a link to the RepRap g-code docs which explains all g-codes in great detail.  Duplicating work like this is not a good use of developers time.

closed 2019-11-03 by klipper-gitissuebot