Klipper3d/klipper #2121: Bltouch clone pin direction reversed

jimfunk, 2019-10-27:
I've pored over the open/closed issues and I can't seem to find anybody having a similar issue.

I had a BLtouch clone (Trianglelab 2019 version) working fine, but suddenly the pin direction reversed. (I don't remember if I did an update just before it. It failed to home properly and broke my XY gantry and it took a while to get the broken parts replaced)

When the printer starts, it does it's self test, and leaves the pin down. If I send `BLTOUCH_DEBUG COMMAND=pin_up` the pin stays down. If I send `pin_down` it brings the pin up. Then if I send `pin_up` it sends the pin down.

This breaks homing completely since it brings the pin up when homing, causing the bed to crash.

My bltouch config looks like this:

    [bltouch]
    sensor_pin: ^ar18
    control_pin: ar11
    pin_up_touch_mode_reports_triggered: False
    x_offset: 39.8
    y_offset: 0
    z_offset: 2.575

Board is an MKS GenL v1.0. The wiring is the same as described at https://www.youtube.com/watch?v=HR-zn4dv1fY

[klippy.log](https://github.com/KevinOConnor/klipper/files/3776596/klippy.log)

KevinOConnor, 2019-10-30:
The attached log file is not complete - we need the full log in order to assist.  Also note that `WARNING:root:No log file specified! Severe timing issues may result!` is a real problem and should be fixed.

That said, this doesn't sound like a klipper issue.  It sounds like the clone bltouch is no longer working correctly.

-Kevin

jimfunk, 2019-10-31:
Here is a log using `--logfile`.

[cube.log](https://github.com/KevinOConnor/klipper/files/3796175/cube.log)

It's very odd since it seems to work fine otherwise. `self_test` and `reset` seem to work.

If I swap the values of `pin_up` and `pin_down` in the `Commands` dict in `bltouch.py` would that screw up homing?

jimfunk, 2019-10-31:
> Here is a log using `--logfile`.
>
> [cube.log](https://github.com/KevinOConnor/klipper/files/3796175/cube.log)
>
> It's very odd since it seems to work fine otherwise. `self_test` and `reset` seem to [...]

Apparently not it seems. The probe deploys and the device lights, but it doesn't trigger when touched. Very odd.

jimfunk, 2019-11-02:
Checked with Marlin. Seems the probe is not working correctly. Sorry for the noise.

closed 2019-11-02 by jimfunk