Klipper3d/klipper #2123: Bad multi mcu connection

albmargar30, 2019-10-28:
Hi, I don't know if this is the right place to comment on this question. I am using a skr V1.3 and a skr mini V1.1 with the example of multi-mcu to control a printer. The movements of X, Y, Z are controlled by skr V1.3 with their respective limit switches, the entire extruder is controlled by skr mini V1.1. Enter the corresponding file klipper.bin on each board when you launch the make menuconfig command. I check that with the command ls / dev / serial / by-path / * it detects the two boards, but when I give it to connect it tells me the printer is not ready, it could be that the mcu of the skr mini V1.1 is giving me problems (STM32F103RCT6)??

albmargar30, 2019-10-28:
[Klipper.log](https://github.com/KevinOConnor/klipper/files/3779681/Klipper.log)

vladbabii, 2019-10-29:
I think it's related to 
- https://github.com/KevinOConnor/klipper/issues/2118
- https://github.com/KevinOConnor/klipper/issues/1013

You need to restart klipper service and power cycle the boards  to get reliable re-connection

albmargar30, 2019-10-29:
Thank you @vladbabii, I have looked at the two conversations and tried to follow the steps you have done with the arduino and the skr mini V1.1, but I cannot get the printer to the ready state. I find it very strange because in another printer I have a 1.3 skr for x, y, z, for the head I use a smoothieboard that both have MCUs of the same family (LPC1768 and LPC1769) and have not given me as much problems as with these two plates. It is possible that being a MCU of different manufacturers costs the firmware more to establish communication. I'm going to pass the log file this morning and the printer.cfg file to see if we find where the fault is.
[Klipper.log](https://github.com/KevinOConnor/klipper/files/3782923/Klipper.log)
[printer.zip](https://github.com/KevinOConnor/klipper/files/3782927/printer.zip)

vladbabii, 2019-10-29:
Try these steps:
* power off boards
* stop klipper -> systemctl stop klipper
* power on boards
* start  klipper -> systemctl start klipper
* use octoprint/dwc to connect to klipper and see status
* use firmware restart once if needed

albmargar30, 2019-10-29:
Is it possible that it is a problem with usb cable? In the other printer the usb cables are very short but in this one, the one that goes from scratch to the skr mini v1.1 that is in the head is 1.5-2 meters. You get a while receiving Send: M105
Recv: ok T: 0, until a while goes by and the value starts getting ready for what happens to ready.

vladbabii, 2019-10-29:
You need to look at klipper logs, not serial output.

albmargar30, 2019-10-30:
I just read the .log file and whenever I establish a connection it says that the printer is not ready, when the time goes by, the message Unable to connect, klipper disconnects and asks me to do firmware_restart, I write in the restart terminal only and the printer It goes to ready state. I do not understand why it does. I attach the .log file to see if someone can see what is happening. I will try to put a shorter cable to see if the problem is the usb cable. The warning I have seen in the .log file are the following
[Klipper.log](https://github.com/KevinOConnor/klipper/files/3787898/Klipper.log)
:

2019-10-30 08:06:12,064 - tornado.access - WARNING - 404 GET /static/webassets/packed_libs.css?ec88168c (::ffff:192.168.1.234) 11.68ms
2019-10-30 08:06:12,102 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.css?6745d981 (::ffff:192.168.1.234) 2.91ms
2019-10-30 08:06:12,112 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.css?0635c0de (::ffff:192.168.1.234) 5.76ms
2019-10-30 08:06:12,118 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.js?7ccee9aa (::ffff:192.168.1.234) 2.73ms
2019-10-30 08:06:12,125 - tornado.access - WARNING - 404 GET /static/webassets/packed_libs.js?ef1e2987 (::ffff:192.168.1.234) 2.67ms
2019-10-30 08:06:12,131 - tornado.access - WARNING - 404 GET /static/webassets/packed_client.js?68a8d0b0 (::ffff:192.168.1.234) 3.08ms
2019-10-30 08:06:12,254 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.js?b00f909b (::ffff:192.168.1.234) 2.63ms
2019-10-30 08:06:14,833 - tornado.access - WARNING - 404 GET /static/webassets/packed_libs.js?ef1e2987 (::ffff:192.168.1.234) 5.46ms
2019-10-30 08:06:14,891 - tornado.access - WARNING - 404 GET /static/webassets/packed_client.js?68a8d0b0 (::ffff:192.168.1.234) 7.73ms
2019-10-30 08:06:14,910 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.js?7ccee9aa (::ffff:192.168.1.234) 5.40ms
2019-10-30 08:06:14,929 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.js?b00f909b (::ffff:192.168.1.234) 5.40ms
2019-10-30 08:06:18,256 - octoprint.server.util.flask - INFO - Passively logging in user VOXELCARE from ::ffff:192.168.1.234
2019-10-30 08:06:18,372 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.css?0635c0de (::ffff:192.168.1.234) 2.69ms
2019-10-30 08:06:18,379 - tornado.access - WARNING - 404 GET /static/webassets/packed_libs.css?ec88168c (::ffff:192.168.1.234) 2.66ms
2019-10-30 08:06:18,385 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.css?6745d981 (::ffff:192.168.1.234) 3.29ms
2019-10-30 08:06:18,504 - tornado.access - WARNING - 404 GET /static/webassets/packed_libs.js?ef1e2987 (::ffff:192.168.1.234) 2.66ms
2019-10-30 08:06:18,511 - tornado.access - WARNING - 404 GET /static/webassets/packed_client.js?68a8d0b0 (::ffff:192.168.1.234) 2.94ms
2019-10-30 08:06:18,517 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.js?7ccee9aa (::ffff:192.168.1.234) 2.61ms
2019-10-30 08:06:18,524 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.js?b00f909b (::ffff:192.168.1.234) 2.93ms
2019-10-30 08:06:18,660 - tornado.access - WARNING - 404 GET /static/webassets/packed_libs.js?ef1e2987 (::ffff:192.168.1.234) 2.68ms
2019-10-30 08:06:18,674 - tornado.access - WARNING - 404 GET /static/webassets/packed_client.js?68a8d0b0 (::ffff:192.168.1.234) 2.63ms
2019-10-30 08:06:18,688 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.js?7ccee9aa (::ffff:192.168.1.234) 2.79ms
2019-10-30 08:06:18,703 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.js?b00f909b (::ffff:192.168.1.234) 2.63ms

BlackStump, 2019-10-30:
you have supplied a Octoprint log  not the klipper log
Please read carefully the below link
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

albmargar30, 2019-10-30:
@BlackStump I am looking for the directory /tmp/klippy.log as it puts the link that you have passed me but I can't find it, I look for it in the windows terminal and I get the following message:

pi@octopi:/home $ /tmp/klippy.log
-bash: /tmp/klippy.log: Permission denied
pi@octopi:/home $

BlackStump, 2019-10-30:
@albmargar30 
If you are using a windows based pc to logon to your pi you need something like this as mentioned in the link
https://winscp.net/eng/docs/introduction

albmargar30, 2019-10-30:
@BlackStump I found the file, I wasn't looking at the root of the system and that's why I couldn't find it. I attach the correct file to see if anyone finds the problem. I have connected the two boards with short usb cables as I have them in another printer and the problem persists.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3788560/klippy.log)

BlackStump, 2019-10-30:
seems you are not making a connection to one of the boards and presume it is the skr mini?
>Unable to open port: [Errno 2] could not open port /dev/serial/by-path/platform-3f980000.usb-usb->0:1.1.2:1.0:

albmargar30, 2019-10-30:
The board that costs to make connection is the skr mini V1.1 (STM32F103RCT6) In another printer I have connected a skr V1.3 and a smoothieboard that carry MCU of the same manufacturer and does not fail, but in this the truth is that It is a despair because I have to wait until the Octoprint terminal shows "Unable to connect" to restart it and finally the skr mini V1.1 connects.

BlackStump, 2019-10-30:
Try adding  restart_method: command  to your config
I am using the skr mini in a 2 mcu config and seems to be ok but had to add the 
restart_method: command
to get it to restart reliably 
ie
>[mcu]
>serial: /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
>restart_method: command

>[mcu hboard]
>serial: /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.2:1.0
>restart_method: command

albmargar30, 2019-10-30:
I have added the "restart_method: command" method as you put it and it works. I wanted to ask if the method has to be taken by both mcu or with only the skr mini V1.1 (mcu hboard) is enough

KevinOConnor, 2019-11-06:
As of PR #2144 you shouldn't need to specify `restart_method: command`.  However, it doesn't hurt to specify it either (on one or both boards).

-Kevin

KevinOConnor, 2019-11-15:
I'm going to close this as there is a good chance the latest code resolves the issue.

-Kevin

closed 2019-11-15 by KevinOConnor