Klipper3d/klipper #2125: Cannot start in simulation mode

spaxxomatic, 2019-10-28:
Hi Kevin
I try to start klipper in simulation mode as per instructions provided in developer info. I use the "linux process" microcontroller architecture, but I get this error during start:

python ./klippy/klippy.py config/avrsim.cfg -i test.gcode -o test.serial -v -d out/klipper.dict
INFO:root:Starting Klippy...
INFO:root:Start printer at Tue Oct 29 00:05:30 2019 (1572303930.8 16638.8)
INFO:root:Extruder max_extrude_ratio=0.103938
INFO:root:Loaded MCU 'mcu' 44 commands (v0.8.0-38-gd65f2e2-dirty-20191029_000213-zero / gcc: (Raspbian 4.9.2-10+deb8u2) 4.9.2 binutils: (GNU Binutils for Raspbian) 2.25)
MCU 'mcu' config: PCA9685_MAX=4096 ADC_MAX=4095 PWM_MAX=32768 CLOCK_FREQ=20000000 MCU=linux STATS_SUMSQ_BASE=256
ERROR:root:Protocol error
Traceback (most recent call last):
  File "./klippy/klippy.py", line 138, in _connect
    cb()
  File "/root/klipper/klippy/mcu.py", line 672, in _connect
    self._check_config()
  File "/root/klipper/klippy/mcu.py", line 613, in _check_config
    self._send_config(None)
  File "/root/klipper/klippy/mcu.py", line 565, in _send_config
    cb()
  File "/root/klipper/klippy/mcu.py", line 276, in _build_config
    "schedule_digital_out oid=%c clock=%u value=%c", cq=cmd_queue)
  File "/root/klipper/klippy/mcu.py", line 716, in lookup_command
    cmd = self._serial.get_msgparser().lookup_command(msgformat)
  File "/root/klipper/klippy/msgproto.py", line 300, in lookup_command
    raise error("Unknown command: %s" % (msgname,))
error: Unknown command: schedule_digital_out

Any hints on why this happens? 
Thanks

KevinOConnor, 2019-10-29:
The linux mcu code doesn't support toggling gpios.  If you compile the host for one of the other targets (eg, avr) then the above commands should be able to produce output.  Further details at: https://www.klipper3d.org/Debugging.html#translating-gcode-files-to-micro-controller-commands

If you want to run the avr micro-controller code in a simulation, then you'll need to download, compile, and run the simulavr program.  Details at: https://www.klipper3d.org/Debugging.html#testing-with-simulavr

-Kevin

KevinOConnor, 2019-11-15:
I'm going to close this as it appears the conversation has concluded.

-Kevin

closed 2019-11-15 by KevinOConnor