Klipper3d/klipper #2126: SPI for serial communication with MCU

NobodyIII, 2019-10-29:
I'm designing a new board to use an FPGA (controlled by an MCU) with Klipper to allow for extremely high step rates with a large number of steppers (up to 12 probably). For communication between Klippy and the MCU, I'd like to use SPI for serial communication, as I'm concerned that UART might be too slow. I see that serial over USB is an option, but I'd like to avoid the need for a USB cable, and instead have the communication over the Raspberry Pi's GPIO pins. SPI looks like a very attractive alternative to UART, considering its speed and its availability on the Pi's GPIO header.

Is this a good idea, or am I missing something? Has any development on this begun yet? If not, how difficult would it be to add this functionality?

NobodyIII, 2019-10-29:
How am I supposed to provide a log when I'm not submitting a bug report?

vladbabii, 2019-10-29:
Just add an empty klippy.log in a zip file.

KevinOConnor, 2019-10-29:
> For communication between Klippy and the MCU, I'd like to use SPI for serial communication, as I'm concerned that UART might be too slow.

Klipper doesn't use a lot of bandwidth.  Even at burst speeds it tends to use no more than about 5-6KB/s.  So, it is only using a fraction of the capacity of regular uart connections.

FWIW, I personally would not recommend SPI nor I2C for board to board communication.  Both protocols are highly susceptible to line noise.  (Uart is equally susceptible to line noise, but Klipper will quickly detect and retransmit in that case. In contrast, line noise on the control lines (clock / chip select) of SPI/I2C can cause the controllers to get into a difficult to clear state.)

That said, if someone wishes to implement support for other forms of communication, then that's fine.

-Kevin

KevinOConnor, 2019-10-29:
> Just add an empty klippy.log in a zip file.

It is not necessary to spoof the bot.  If a ticket doesn't "reference an event that has occurred while running the software" then I will clear the "incomplete issue" label from it.

-Kevin

sensille, 2019-10-29:
Hi @NobodyIII,

I'm also working on an FPGA-based board and already have some code in place. Maybe you'd like to join #klipper in IRC and we can discuss if it makes sense to join forces or share experiences.

-sensille

NobodyIII, 2019-11-03:
Thanks for the quick responses. I'll stick to UART. Also, I'll try discussing the FPGA board with @sensille over IRC.

closed 2019-11-03 by NobodyIII