Klipper3d/klipper #2127: Cannot compile firmware for simulavr

vladbabii, 2019-10-29:
Did the steps described in documentation
```
...compile Klipper for use in simulavr, run:

cd /patch/to/klipper
make menuconfig
and compile the micro-controller software for an AVR atmega644p, set the MCU frequency to 20Mhz, and select SIMULAVR software emulation support. Then one can compile Klipper (run make) ...
```

result of make:
```
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
src/sched.c:10:31: fatal error: board/io.h: No such file or directory
compilation terminated.
Makefile:66: recipe for target 'out/src/sched.o' failed
make: *** [out/src/sched.o] Error 1
```

This is on a fresh clone of klipper from github - d65f2e2

.config file from klipper root folder
[dot_config.zip](https://github.com/KevinOConnor/klipper/files/3782810/dot_config.zip)

empty log for bot, please ignore
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3782811/klippy.zip)

KevinOConnor, 2019-10-30:
Works fine for me.  You probably need to run `make clean` first.

-Kevin

vladbabii, 2019-11-01:
Re installed klipper on a fresh install and it worked.

closed 2019-11-01 by vladbabii