Klipper3d/klipper #2131: Unnecessary After in systemd unit file

mrvanes, 2019-10-30:
https://github.com/KevinOConnor/klipper/blob/d65f2e28612d366bd06c63902b2f1bfcbc06bd37/scripts/install-ubuntu-18.04.sh#L55

Why should Klipper start after network? Octoprint should start after network and Klipper, but Klipper can't be started early enough I think?

KevinOConnor, 2019-11-05:
Perhaps @nathantsoi has a comment on this.

-Kevin

nathantsoi, 2019-11-05:
yep, pry an artifact of me copying this from something else. feel free to remove

KevinOConnor, 2019-11-22:
Okay, I'm going to close this ticket as it appears the original question was answered.  If a change is desired, then a developer will need to submit a pull request with that change.

-Kevin

closed 2019-11-22 by KevinOConnor

yodor, 2019-11-28:
Hello all,
I was looking for a way to trigger klipper state following usb printer power on/off.
This is actually following the state of the USB port used on the system.
'BindsTo' takes care to stop klipper service on power off or unplug
'After' and 'Requisite' to start 
To find the correct device plug your printer and power it on then you can use 
systemctl --all --full -t device

#Systemd service file for klipper
[Unit]
Description=Klipper Service
After=dev-serial-by\x2did-usb\x2dRRD__www.ru_RUMBA_\x2d_ATmega_2560_co_557323238303515092E0\x2dif00.device
BindsTo=dev-serial-by\x2did-usb\x2dRRD__www.ru_RUMBA_\x2d_ATmega_2560_co_557323238303515092E0\x2dif00.device
Requisite=dev-serial-by\x2did-usb\x2dRRD__www.ru_RUMBA_\x2d_ATmega_2560_co_557323238303515092E0\x2dif00.device

[Install]
WantedBy=dev-serial-by\x2did-usb\x2dRRD__www.ru_RUMBA_\x2d_ATmega_2560_co_557323238303515092E0\x2dif00.device

[Service]
Type=simple
User=klipper
RemainAfterExit=yes
ExecStart=/home/klipper/venv/bin/python /home/klipper/src/klipper/klippy/klippy.py -v -l /tmp/klipper.log /home/klipper/config-k218.cfg