Klipper3d/klipper #2132: Reported error: Heater extruder not heating at expected rate

dcv10, 2019-10-30:
I am continuously experiencing the same issue: "Your printer's firmware reported an error. Due to that, the ongoing print job will be canceled. Reported error: Heater extruder not heating at expected rate" while trying to complete a 15hr print. This happens about 3 minutes into the print. 

I replaced the thermistor, set the max_power for the extruder to 0.8, max_power for heated bed to 0.3, added smooth_line, done PID tuning and the problem still persists. 

However, when the printer is stationary, the temperature stays constant.

[octoprint.log](https://github.com/KevinOConnor/klipper/files/3787859/octoprint.log)

FHeilmann, 2019-10-30:
Two suggestions:

Do you have a cooling fan that kicks on after some time of printing? The resulting cooldown of the nozzle can trigger this thermal shutdown. 

Looking at your log your hotend temperatur drops from 205 to 194 degreees within 1 second, which can be the result of faulty wiring (i.e. wiring that has a break which momentarily disconnects the thermistor as the printer moves). Since you already replaced the thermistor the break is probably somewhere in the wire going from your thermistor to your 3D printer controller.

See of one of those two things helps you out.

dcv10, 2019-11-04:
@FHeilmann Thanks for your help. You were correct, it was the wiring.

closed 2019-11-04 by dcv10