Klipper3d/klipper #2133: bed_mesh is not taken into account for delta printer

dzervas, 2019-10-30:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3789780/klippy.log)
[rossel.txt](https://github.com/KevinOConnor/klipper/files/3789781/rossel.txt)

I have a delta printer and the bed is tilted for sure (probably not warped, as I have glass on top of it). The problem is that even after delta_calibrate and bed_mesh_calibrate, when I try to print the bed_mesh is not taken into account. it's too high on y and too low on -y. I have re-measured the x and y offset of the probe (so that the bed mesh is not "shifted"). The z offset doesn't seem to be a problem for now. I've attached the log and my current config.

Arksine, 2019-11-01:
You say that the bed is tilted, not warped, however the points in your saved "bed_mesh default" profile do not reflect that.  The results are quite strange, generally the center of the bed is highest, outside of a small section as you move toward -x,-y, where the bed is higher.  Generally the rest of the bed slopes down from the center.

Unless you are clearing the mesh prior to a print then it is being applied, you can verify by moving the tool to different x,y coordinates and executing the GET_POSITION gcode, noting the change to the toolhead's z position.  After verifying that the toolhead's z position is changing, you know that the mesh is being applied.   This will tell you that the probed points aren't accurately measuring the shape of the bed.  That narrows it down to an issue with the probe itself, a mechanical issue, or delta calibration being off.

dzervas, 2019-11-01:
hmmm. ok, I'll try manual calibration...

dzervas, 2019-11-01:
Yeap, my probe is wildy inaccurate...
For anyone with the same problem:
I've got a bltouch clone (3dtouch) and
1. it was too high and I tried to lower it with a couple of nuts (wrong)
2. my effector was upside down (e3d v6 effector), that's why the probe was so high (lol)
3. the pin was too low, so I lose accuracy (at this point I'm ~0.1mm accuracy)
4. The e3d v6 fan was resonating and made the probe inaccurate. I'm at ~0.4mm accuracy right now

with some more tinkering (speeds, times, etc.) I should be at 0.25mm or better.
Once again the issue was mechanical.. I'm very sorry klipper team...

closed 2019-11-01 by dzervas