Klipper3d/klipper #2134: Move exceeds maximum extrusion during filament change

brycedjohnson, 2019-10-30:
Set up filament sensor and script.  I've had it working a few times, but sometimes I get that move exceeds maximum extrusion.  Been trying different things out with trial and error, but haven't been successful yet.  Any input on what the issue could be?

```Move exceeds maximum extrusion (5.477mm^2 vs 0.640mm^2)```

[klippy.log](https://github.com/KevinOConnor/klipper/files/3791227/klippy.log)

KevinOConnor, 2019-11-06:
You should have received another message immediately after the first: `See the 'max_extrude_cross_section' config option for details` .  That option is described in config/example.cfg.

This error is usually the result of a "priming line" at the start of a print that triggers the check (my advice would be to change the priming procedure then), or due to a slicer using some form of "extruder pressure adjustment" - for the latter see https://www.klipper3d.org/Slicers.html for some hints to disable.

-Kevin

KevinOConnor, 2019-11-22:
I'm going to close this ticket as it appears the conversation has concluded.

-Kevin

closed 2019-11-22 by KevinOConnor

brycedjohnson, 2019-12-09:
Thanks @KevinOConnor, I meant to follow up.  I ran into this when I was telling octoprint extrude after a filament change.  I increased max_extrude values to
```
max_extrude_only_velocity: 120
max_extrude_only_accel: 6400
```
and now I don't get the error.