Klipper3d/klipper #2138: Not ADC five buttons

amaix1086, 2019-11-03:
Hi. There is a printer from the Korean company Rokit 3dison AEP, (like a makerbot but modified). The question is to connect its buttons - it is not ADC like on the anet A8 but it is on digital pins.
There are only four buttons in the anet A8 configs, although there are actually five of them. 
Is it possible to use this method of connecting buttons, or only ADC buttons are available?

KevinOConnor, 2019-11-05:
It may be possible, but I'd guess it would require a schematic of how those pins are wired along with a developer to implement it (if it's not something already implemented).

-Kevin

KevinOConnor, 2019-11-22:
I'm going to close this ticket as it appears the conversation has concluded.

-Kevin

closed 2019-11-22 by KevinOConnor