Klipper3d/klipper #2139: Move exceeds maximum extrusion (0.656mm^2 vs 0.640mm^2)

LegendaryCheez, 2019-11-03:
Hello, I have been hitting an issue with printing on my stock Ender 3 since I have installed Klipper following Teaching Tech's Youtube video:  https://www.youtube.com/watch?v=yAfalR7-Tvw

Every time that I go to print, I get the following error.  I haven't been able to find the solution for this, but it looks like the error kicks off on the first extrusion of filament.  I have tried resetting my starting GCode in Cura and that did not fix the problem.  Attached is the log and the GCode for the print.  Thanks for your help!
  
[klippy.log](https://github.com/KevinOConnor/klipper/files/3801824/klippy.log)
[CE3_xyzCalibration_cube.gcode.txt](https://github.com/KevinOConnor/klipper/files/3801825/CE3_xyzCalibration_cube.gcode.txt)

gdachs, 2019-11-03:
Just remove this from the start GCODE and it should work:
`G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.5 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.5 Y75.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.8 Y75.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.8 Y20 Z0.3 F1500.0 E30 ; Draw the second line`

LegendaryCheez, 2019-11-03:
Print started!  I could have swore I tried removing the starting GCode, but I guess I didn't.  Thank you for the quick response and the help!

closed 2019-11-03 by LegendaryCheez