Klipper3d/klipper #2140: Manual steppers with [tmc2209 stepper_name] not supported

vladbabii, 2019-11-03:
I use a manual stepper for my toolchanger locking mechanism. Upgraded today to uart 2209 stepper but it does not work.

Error:
```
Config error
Traceback (most recent call last):
  File "/opt/klipper/klippy/klippy.py", line 134, in _connect
    self._read_config()
  File "/opt/klipper/klippy/klippy.py", line 127, in _read_config
    self.try_load_module(config, section_config.get_name())
  File "/opt/klipper/klippy/klippy.py", line 116, in try_load_module
    self.objects[section] = init_func(config.getsection(section))
  File "/opt/klipper/klippy/extras/tmc2209.py", line 95, in load_config_prefix
    return TMC2209(config)
  File "/opt/klipper/klippy/extras/tmc2209.py", line 76, in __init__
    tmc.TMCStealthchopHelper(config, self.mcu_tmc, TMC_FREQUENCY)
  File "/opt/klipper/klippy/extras/tmc.py", line 287, in TMCStealthchopHelper
    step_dist = stepper_config.getfloat('step_distance')
  File "/opt/klipper/klippy/configfile.py", line 60, in getfloat
    minval, maxval, above, below)
  File "/opt/klipper/klippy/configfile.py", line 31, in _get_wrapper
    v = parser(self.section, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 362, in getfloat
    return self._get(section, float, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 356, in _get
    return conv(self.get(section, option))
  File "/usr/lib/python2.7/ConfigParser.py", line 330, in get
    raise NoSectionError(section)
NoSectionError: No section: 'lock_stepper'
No section: 'lock_stepper'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
```

Config:
```
[manual_stepper lock_stepper]
step_pin: xy:PB10
dir_pin: xy:PB11
enable_pin: !xy:PB2
step_distance: .1
endstop_pin: !xy:PA2

[tmc2209 lock_stepper]
uart_pin: xy:PB7
microsteps: 16
interpolate: True
run_current: 0.9
stealthchop_threshold: 50
```

Log:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3802193/klippy.zip)

Thank you!

KevinOConnor, 2019-11-05:
Use the full config section name: `[tmc2209 manual_stepper lock_stepper]`

-Kevin

vladbabii, 2019-11-05:
since 
```
DUMP_TMC STEPPER="lock_stepper"
```
works, i was expecting it to be the way i tried it. Maybe it's just me but it's a bit confusing configuration requires full name and command doesn't

Can you please add this to the documentation as an example ?
If you don't have time i will create a pull request, I'd like to add a comment next to manual stepper to describe the [tmcX manual_stepper Y] part and that commands require only the name.

Thank you for the clarification.

vladbabii, 2019-11-07:
@KevinOConnor i will close this since it's a non-issue. I will try to do a pull a pull request for documentation as soon as possible.

closed 2019-11-07 by vladbabii