Klipper3d/klipper #2143: Unable to calibrate EZABL probe on CR-10S

boardthatpowder, 2019-11-05:
Attempting to set up Klipper for a CR-10S that's has an EZABL probe.  Got to the point where I need to set my z-offset.  Following the instructions, I'm trying to run `PROBE_CALIBRATE` but keep encountering the following error in the logs...

```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 173, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 250, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 277, in _dispatch_loop
    timeout = self._check_timers(eventtime)
  File "/home/pi/klipper/klippy/reactor.py", line 135, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/reactor.py", line 47, in invoke
    res = self.callback(eventtime)
  File "/home/pi/klipper/klippy/mcu.py", line 210, in _home_retry
    self._trigger_notify()
  File "/home/pi/klipper/klippy/homing.py", line 37, in _endstop_notify
    self.toolhead.signal_drip_mode_end()
  File "/home/pi/klipper/klippy/toolhead.py", line 461, in signal_drip_mode_end
    self.drip_completion.complete(True)
AttributeError: 'NoneType' object has no attribute 'complete'
```

Any help would be most appreciated.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3807033/klippy.log)

KevinOConnor, 2019-11-06:
Not sure what went wrong.  Could you retry, issue M112 immediately after the error, and then attach that Klipper log?

-Kevin

boardthatpowder, 2019-11-07:
Just tried this again, and now the PROBE_CALIBRATE function is working as expected.  Will close the issue.  Thanks!

closed 2019-11-07 by boardthatpowder