Klipper3d/klipper #2145: [Feature Request] Multiple Motors on Extruder - Push / Pull Filament from large spool

drphil3d, 2019-11-05:
I'd love to see a feature that would make it possible to have two stepper motors on a single extruder that run at the same time, with different step rate. 
This would make it possible to have a lower end extruder pulling filament from a large spool and a higher end extruder driving the hotend. I don't currently see any way to easily and accurately do this.

Thanks!

jalanjarosz, 2019-11-05:
This request looks to be the same as in: https://github.com/KevinOConnor/klipper/issues/1267

drphil3d, 2019-11-06:
Ah, I see. I think I will wait for it to be supported in the main branch sense the fork it was built in is 0.7 and were at 0.8 so I would lose other features I commonly use.

KevinOConnor, 2019-11-15:
I'm going to close this in favour of the existing #1267 .

-Kevin

closed 2019-11-15 by KevinOConnor