Klipper3d/klipper #2146: Problem with manual stepper (case-sensitivity, no response?)

Spacemarine2018, 2019-11-05:
My config: SKR Pro Board from Bigtreetech with TMC5160 V1.2 from Bigtreetech. Raspberry Pi 4 B running Duet Web Control

I want to use a manual stepper for cutting filament, so I defined it as:

[manual_stepper cutter]
step_pin = PD15
dir_pin = PE7
enable_pin = !PA3
step_distance = .003125
velocity = 70
accel = 1200

When I issue "MANUAL_STEPPER STEPPER=CUTTER ENABLE=1 MOVE=10"
the response is "failed: MANUAL_STEPPERThe value 'CUTTER' is not valid for STEPPER"

It makes no difference if I issue CUTTER or cutter on the gcode console, the output is the same and always uppercase.

Then I changed cutter to CUTTER in the printer.cfg, so it is now:

[manual_stepper CUTTER]
step_pin = PD15
dir_pin = PE7
enable_pin = !PA3
step_distance = .003125
velocity = 70
accel = 1200

When I run "MANUAL_STEPPER STEPPER=CUTTER ENABLE=1 MOVE=10" the g-code console get's stuck, as it is waiting for a return message that never comes. Also there is no movement from the stepper.

All steps can be seen in the klippy.log, I did a fresh restart before the experiment to get a clean log that shows only the behaviour described above.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3807801/klippy.log)

Spacemarine2018, 2019-11-05:
I'm not really sure what you mean by that? I installed Klipper as instructed, I didn't change anything in the Klipper code. I made only changes to printer.cfg, nothing else.

KevinOConnor, 2019-11-05:
The dwc code alters the internal state of Klipper - it is not supported here.

-Kevin

vladbabii, 2019-11-06:
@Spacemarine2018 octoprint doesn't, you can test with it safely

Spacemarine2018, 2019-11-10:
Indeed, once I switched to Octoprint, the issue was gone. It seems DWC
somehow tinkers with upper-/lowercase of the commands, Octoprint does not. I will use Octoprint from now on.

I'll report the problem on the DWC bugtracker.

Maybe one final hint for people searching how to operate manual steppers, as this gave me a lot of headdache and was non-obvious to me and others even after reading the docs extensively:

The correct syntax for manual steppers is:
[manual_stepper cutter]
[tmc5160 manual_stepper cutter]

and NOT as some people are led to believe: [tmc5160 cutter]

closed 2019-11-10 by Spacemarine2018