Klipper3d/klipper #2148: Klipper reports temp 0 until restarted

mrvanes, 2019-11-05:
On cold boot, klippy continously reports tool and bed temp 0, until I give a restart in terminal windows of octoprint:
```
Send: M105
Recv: ok B:0.0 /0.0 T0:0.0 /0.0
Send: M105
Recv: ok B:0.0 /0.0 T0:0.0 /0.0
Send: restart
Recv: ok
Recv: // Klipper state: Disconnect
Send: M105
Recv: ok T:0
Recv: // Klipper state: Ready
Send: M105
Recv: ok B:20.2 /0.0 T0:20.0 /0.0
Send: M105
Recv: ok B:20.3 /0.0 T0:20.0 /0.0
```
Attached is the klippy log including the restart command
[klippy.log](https://github.com/KevinOConnor/klipper/files/3810706/klippy.log)

KevinOConnor, 2019-11-06:
The log indicates a `Timeout with MCU 'mcu' ` error.  If this is just a one off, then nothing to worry about.  If it reoccurs, the best advice we have is at: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

KevinOConnor, 2019-11-06:
Oh, it looks like you've done a custom install of the software.  If this problem occurs reliably on a reboot, then check for ModemManager (or similar) on the host machine - it's known to cause issues like the above.

-Kevin

mrvanes, 2019-11-07:
Removed modemmanager, problem gone. Thanks!!

closed 2019-11-07 by mrvanes