Klipper3d/klipper #2149: Feature Request: Progressive fade of mesh correction

mattkenn4545, 2019-11-06:
When printing at very fine layer heights I get surface artifacts that only appear when bed mesh correction is active. 

My bed is fairly level, < 0.3mm over the entire bed, however that is still too much to get a consistent first layer/good adhesion without mesh correction.

What would be great if mesh correction could progressively fade to 0 over the first several layers and/or up-to a configurable z-height. This would lead to layer inconsistency for those initial layers but would avoid micro-stepping across most layers when using 'magic number' layer heights.

FHeilmann, 2019-11-06:
You can already do that using `fade_start` and `fade_end` in your config:

Here's the relevant part of the docs: https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L73

mattkenn4545, 2019-11-06:
Awsome! closing

closed 2019-11-06 by mattkenn4545