Klipper3d/klipper #2150: Request: Other "support" channel than IRC or mailing list

ProfDrYoMan, 2019-11-06:
Hi,

I know there is a low noise mailing list and an IRC channel listed in the documentation.

Is there any chance to establish a less "geeky" "support" channel than IRC?

Let's say a free tier slack or discord?

Or is there any other "support" channel I missed?

Cu.

theopensourcerer, 2019-11-06:
There is a fairly active group on Facebook with >4,000 members:

https://www.facebook.com/groups/klipper/

mrvanes, 2019-11-06:
The discord server is very active as well
https://discordapp.com/channels/431557959978450984/513935277723746314

ProfDrYoMan, 2019-11-06:
> There is a fairly active group on Facebook with >4,000 members:
> 
> https://www.facebook.com/groups/klipper/

Really? FB?

ProfDrYoMan, 2019-11-06:
> The discord server is very active as well
> https://discordapp.com/channels/431557959978450984/513935277723746314

This link does not get me to a channel. Access needed?

mrvanes, 2019-11-06:
https://www.freelists.org/post/klipper/Discord-Server,1

KevinOConnor, 2019-11-07:
The only official Klipper communication channels are described at: https://www.klipper3d.org/Contact.html

Introducing another channel (slack, discord, etc.) has been discussed in the past.  However, I personally, don't have any short term plans to add another official channel currently.

-Kevin

Lenbok, 2019-11-07:
Is IRC really that much more geeky than slack/discord? Maybe the help page could point people who don't have an IRC client to a browser interface like https://webchat.freenode.net/

vladbabii, 2019-11-07:
I currently am more active on the unofficial discord due to better history and cross-platform clients. The android irc clients seem clumsy at best. 

On discord we make sure to specify that we are unofficial help and we direct people to irc and github for official stuff - we're just enthusiast helping each other.

@ProfDrYoMan - heres a freshly generated https://discord.gg/ZPg3fYF

ProfDrYoMan, 2019-11-07:
> Is IRC really that much more geeky than slack/discord? Maybe the help page could point people who don't have an IRC client to a browser interface like https://webchat.freenode.net/

Slack/Discord, because:

- Full server side logging which is searchable.
- Easy attachment of pictures, videos, configfiles. (Searchable).
- Different easy to spot channels for different topics.
- Bundling of communications into threads if wanted.
- ...

Lot more then possible in IRC.

And talking of FB. You cannot even search in FB groups. Or at least it sucks. => FB groups are bad in keeping memory.

ProfDrYoMan, 2019-11-07:
> @ProfDrYoMan - heres a freshly generated https://discord.gg/ZPg3fYF

Thanks. Got access there.

Yes, I fully agree with you, discord suits much better that IRC in any means.

But if Kevin doesn't want to got that way these will stay unofficial and the only way to get direct contact to him will be IRC/ML or direct mail.

KevinOConnor, 2019-12-01:
Thanks for the feedback.  It looks like the conversation has concluded on this, so I'm going to close this issue.

-Kevin

closed 2019-12-01 by KevinOConnor