Klipper3d/klipper #2151: Full delta calibration process?

dingbatca, 2019-11-06:
Trying to understand the full process for calibrating a Delta.  This is going from a clean config file, like mine below, to a fully calibrated and leveled Delta.

This is what my process looks like, but it falls apart early on, and I would like some help better understanding the order of operations and what I should do differently.

Heat everything up, and let it stabilize 
[Calibrate my Z Probe](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md)
[Basic delta calibrate](https://github.com/KevinOConnor/klipper/blob/master/docs/Delta_Calibrate.md)
And here is what everything falls apart.  I want to do the extended calibration, but cant because the bed is not level enough.  When I try leveling the bed (BED_MESH_CALIBRATE), its still not level.  In some spots the tool head hits the deck, and others it's almost 1mm off.  My bed is borosilicate glass, so I am rather sure it's relatively consistent.  Z Probe is a [mini differential ir height sensor](https://www.filastruder.com/products/mini-differential-ir-height-sensor).

I am stuck.  Am I using the wrong calibration sequence, bed leveling settings, general printer settings?  What should I look at/fix/change?

```
Start printer at Wed Nov  6 08:24:00 2019 (1573057441.0 157267.4)
===== Config file =====
[stepper_a]
step_pin = PE9
dir_pin = PF1
enable_pin = !PF2
step_distance = 0.005
endstop_pin = !PB10
position_endstop = 528.58
homing_speed = 50
arm_length = 304.11

[tmc5160 stepper_a]
cs_pin = PA15
spi_software_miso_pin = PC11
spi_software_mosi_pin = PC12
spi_software_sclk_pin = PC10
microsteps = 32
interpolate = True
run_current = 0.8
hold_current = 0.4
sense_resistor = 0.075

[stepper_b]
step_pin = PE11
dir_pin = PE8
enable_pin = !PD7
step_distance = 0.005
endstop_pin = !PE12

[tmc5160 stepper_b]
cs_pin = PB8
spi_software_miso_pin = PC11
spi_software_mosi_pin = PC12
spi_software_sclk_pin = PC10
microsteps = 32
interpolate = True
run_current = 0.8
hold_current = 0.4
sense_resistor = 0.075

[stepper_c]
step_pin = PE13
dir_pin = PC2
enable_pin = !PC0
step_distance = 0.005
endstop_pin = !PG8

[tmc5160 stepper_c]
cs_pin = PB9
spi_software_miso_pin = PC11
spi_software_mosi_pin = PC12
spi_software_sclk_pin = PC10
microsteps = 32
interpolate = True
run_current = 0.8
hold_current = 0.4
sense_resistor = 0.075

[extruder]
step_pin = PE14
dir_pin = PA0
enable_pin = !PC3
step_distance = 0.00151125925
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PB1
control = pid
pid_kp = 28.3
pid_ki = 1.526
pid_kd = 131
min_temp = 10
min_extrude_temp = 10
max_temp = 300
max_power = 1
sensor_type = MAX31865
spi_software_sclk_pin = PD0
spi_software_mosi_pin = PE4
spi_software_miso_pin = PD2
sensor_pin = PE2
rtd_reference_r = 430
rtd_nominal_r = 100
rtd_num_of_wires = 3
rtd_use_50hz_filter = True

[tmc5160 extruder]
cs_pin = PB3
spi_software_miso_pin = PC11
spi_software_mosi_pin = PC12
spi_software_sclk_pin = PC10
microsteps = 4
interpolate = True
run_current = 0.8
hold_current = 0.4
sense_resistor = 0.075

[heater_bed]
heater_pin = PD12
sensor_pin = PF3
sensor_type = ATC Semitec 104GT-2
control = pid
pid_kp = 108.6
pid_ki = 3.825
pid_kd = 771
min_temp = 0
max_temp = 120

[fan]
pin = PE5
max_power = 1
kick_start_time = 1.000

[heater_fan fan1]
pin = PC8
max_power = 1
heater = extruder

[controller_fan fan2]
pin = PE6
max_power = 1
idle_timeout = 15
idle_speed = 0.75
heater = extruder

[mcu]
serial = /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00

[probe]
pin = PE15
z_offset = -0.062
y_offset = 40.00
speed = 5
samples = 3
sample_retract_dist = 3.1416

[printer]
kinematics = delta
max_velocity = 1000
max_accel = 3000
max_z_velocity = 1000
minimum_z_position = -5
delta_radius = 178.54

[delta_calibrate]
radius = 130
speed = 50

[bed_mesh]
speed = 50
min_point = -85,-85
max_point = 85,85
probe_count = 3,3

[board_pins]
aliases =
        EXP1_1=PG4, EXP1_3=PD11, EXP1_5=PG2, EXP1_7=PG6, EXP1_9=<GND>,
        EXP1_2=PA8, EXP1_4=PD10, EXP1_6=PG3, EXP1_8=PG7, EXP1_10=<5V>,
        EXP2_1=PB14, EXP2_3=PG10, EXP2_5=PF11, EXP2_7=PF12,  EXP2_9=<GND>,
        EXP2_2=PB13, EXP2_4=PB12, EXP2_6=PB15, EXP2_8=<RST>, EXP2_10=PF13
=======================
```

dingbatca, 2019-11-06:
Attached is my log.  Not very exciting.
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/3815921/klippy.log.gz)

dingbatca, 2019-11-06:
I guess klipper-gitissuebot does not like .gz.   Attached is my .log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3815973/klippy.log)

BlackStump, 2019-11-06:
A critical factor for any delta calibration process is having a ideally perfectly flat, level and square to the towers bed.
any calibration software expects that so if you don,t have (which most don,t including me) then the software calibration will have problems, this is not a klipper specific problem it is a Delta problem in general. 
If you have a probe that is offset from the nozzle then you may well have effector tilt problem that will affect the calibration problem.
I would suggest manually calibrating the delta.

The Problem, one has a tilted bed or one that is not flat so one thinks ok I will use bed mesh to flatten and square up the bed but the above comments prevent that because you need a perfectly calibrated delta before you can use bed mesh on a delta.

ir sensors for which I have tried on a delta do suffer from light reflection but effector tilt can be a bigger problem unless your delta build is perfect.
>Glass (with or without coatings such as hairspray, PVA or Kapton tape): works as-is if placed directly >on a PCB bed heater or other surface that does not reflect strongly. If there is an **aluminium heat** >spreader or bed plate underneath the glass, then either paint the aluminium surface matt black (see >below), or put a sheet of matt black paper between the glass and the aluminium. Coatings on the >glass affect the trigger height slightly.

precisely measuring your diagonal rods does help as it will be one parameter you can use that is correct and precise. one less parameter to deal with.

while software calibration on a very well built delta can be a joy to use, on any delta less then perfect it can have you going around in circles imho.
Don,t attempt a software calibration imho until you have manually calibrated it first, record the manual calibration parameters ie endstops and delta radius if software calibration differs a lot from a manual calibration you know the software is wrong because of all of the above reasons.

dingbatca, 2019-11-06:
@BlackStump 
Ya...  I was afraid some one was going to say that.  Regardless of how hard I try, I am still only human. I have spent 3 months trying to get this Delta to fly and just having no luck.  Multiple firmware's, 3 different controller boards, a bunch of different sensors (Z/endstop)... 

The learning has been worth it, but I was kinda hoping for a running machine at the end...

KevinOConnor, 2019-11-07:
> Heat everything up, and let it stabilize
Calibrate my Z Probe
Basic delta calibrate
And here is what everything falls apart. I want to do the extended calibration, but cant because the bed is not level enough. 

I recommend following the directions at https://www.klipper3d.org/Delta_Calibrate.html to perform the manual probing steps (`DELTA_CALIBRATE METHOD=manual`).  I too have a delta printer that came with an automatic probe and the probe is useless.  Even a tiny amount of effector tilt makes the probe completely inaccurate.  It only takes a couple of minutes to perform manual delta probing.  My printer works fine once calibrated using the steps described at the link above.

-Kevin

BlackStump, 2019-11-07:
I totally agree with Kevin,s recommedation just do the Manual Calibration as described

BlackStump, 2019-11-07:
A perfect flat bed helps all printers but with delta,s and Automatic Probes it is a requirement imho.
The best probe is a probe that uses the nozzle as the probe, the one that comes to mind is a piezo.

dingbatca, 2019-11-08:
I got over my pity party about my Delta and got back to work.  Spent better then 4 hours clubbing her with the calipers last night. NO automatic, or firmware calibration.  Once I got all that done I was able to do the basic, and the extended calibration.

Is this the correct order of operations?
Heat up and let stabilize 
Basic delta calibrate, manual mode
Extended mode
Bed leveling?

Any tricks on setting up bed leveling for a Delta (round bed)?  Here is my current bed_mesh settings:
```
[bed_mesh]
speed = 50
min_point = -85,-85
max_point = 85,85
probe_count = 3,3
```

theccm, 2019-11-10:
You don’t want to do this IMO : Heat up and let stabilize
When you do the manual calibration process, you have to do it with a cold extruder. We use the sheet paper test to compensate the  extruder expansion when heating it up. See the guide which explain it.

KevinOConnor, 2019-12-01:
I'm going to close this issue as it appears a solution was found for the original issue.

-Kevin

closed 2019-12-01 by KevinOConnor