Klipper3d/klipper #2152: default_parameter_xyz not used to template macros

wizhippo, 2019-11-07:
If a macro is called with no params but default_parameter_X exist the params.X is not set

KevinOConnor, 2019-11-07:
That is by design - the `params` variable is intended to provide the actual variables on the command line.  Use just `X` to get the combination of default_parameter_X or params.X.

-Kevin

wizhippo, 2019-11-07:
Will do.  Didn't realize could use just x

wizhippo, 2019-11-07:
Does not work for me.  I get `!! Error evaluating 'gcode_macro FILAMENT_UNLOAD:gcode': TypeError: an integer is required`

```
[gcode_macro FILAMENT_LOAD]
default_parameter_FAST: 13
default_parameter_SLOW: 1
gcode:
  M83
  {% for count in range(FAST) %}
  G1 E50 F1800
  {% endfor %}
  {% for count in range(SLOW) %}
  G1 E50 F200
  {% endfor %}
  M82
```

wizhippo, 2019-11-07:
never mind. forgot to cat to int

closed 2019-11-07 by wizhippo