Klipper3d/klipper #2154: Feature Request: Open a socket (not only serial /tmp/printer) for communication

ProfDrYoMan, 2019-11-07:
See
https://github.com/intelligent-agent/redeem/blob/904d1588056bb05f1df2d039fd3d5d92f45210d4/redeem/Ethernet.py

In need to separate OctoPrint from Klipper on my BBB because the BBB has not enough processing power for both to get stutter free prints.

I do so by doing
`socat -d -d open:/tmp/printer tcp-l:22113,nodelay`
on the BBB and
`socat -d -d pty,wait-slave,raw,echo=0,link=./klipper tcp:beaglebone:22113`
on the machine with OctoPrint.

That way I can use the ./klipper pipe in OctoPrint to access Klipper on the other machine.
Works flawless.

In redeem (replicape printer firmware) I could skip the first part, since redem provides an network link on port 50000.

So
`socat -d -d pty,wait-slave,raw,echo=0,link=./klipper tcp:beaglebone:50000`
on the machine with OctoPrint was sufficent.

If we get some network connectivity into Klipper I will try to get rid of the socat for OctoPrint by trying get that included there also.

jakep82, 2019-11-07:
I"m not sure what is being requested here, but I can say that Octoprint and Klipper work fine on a BBB as long as you use virtual SD.  I've made many prints on my Replicape using this setup.

KevinOConnor, 2019-11-07:
Looks like a dup of #1097 .

-Kevin

ProfDrYoMan, 2019-11-07:
Hi Kevin.

Yes, #1097 is the same.

The link to the code in redeem gives a pretty good view on how to implement, I guess.

KevinOConnor, 2019-11-15:
I'm going to close this in favour of the existing #1097 .

-Kevin

closed 2019-11-15 by KevinOConnor