Klipper3d/klipper #2155: Feature request: different acceleration limits for X and Y in CoreXY

Spacemarine2018, 2019-11-07:
CoreXY machines typically have different inertias for X and Y, so it would be desireable to set different acceleration limits for X and Y. At the moment only one limit for the whole machine can be set. Cartesion architectures could also benefit from that, but not as much since one can use pulleys of different diameter for X and Y to compensate for differences in inertia.

I'm building a high speed pen-plotter based on CoreXY and Y is about 10 times as heavy as X, and I'd like to accelerate X much faster than Y. The g-code I'm using does not contain any speed parameters, so the machine always operates at the global acceleration and speed limit.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3819744/klippy.log)

KevinOConnor, 2019-11-07:
Looks like a dup of #679.

-Kevin

KevinOConnor, 2019-11-15:
I'm going to close this in favour of the existing #256 .

-Kevin

closed 2019-11-15 by KevinOConnor