Klipper3d/klipper #2158: Fan problem MKS Gen 1.4

Matrixer007, 2019-11-07:
Hi!
First, I apologize for my bad English.
With my CR-20 with MKS Gen 1.4 on Raspberri Pi 3+ I have a problem with the printpart cooler. It does not matter if I insert M106 S255 or M106 S1, the fan runs full. M106 S0 or M107 stops it normally. After some experiments, I have found that I can control the fan as desired with values between 0.1 and 1. Practically, the values are shifted by a factor of 100, so if I enter M106 S0.5, for example, the fan runs at 50%. M106 S70 gives 70% and so on. Unfortunately, no values below 1 can be set in the slicer. I already tried with hardware_pwm: False and True, no change. Even the reinstall of Octopi and Klipper did not succeed.
With Marlin there are no problems, the fan can be controlled normally.
Best regards 
Uwe

[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/3820939/klippy.log.txt)

KevinOConnor, 2019-11-08:
You've set the kick_start_time to 5 seconds, which seems incorrect and would complicate debugging.  Remove that setting and retry.

If you still have issues, them run the command you are having trouble with (eg, `M106 S100`), then issue an `M112`, and then attach that Klipper log file here.  See https://www.klipper3d.org/Contact.html for further details.

-Kevin

Matrixer007, 2019-11-09:
Thanks for answer!
I tried it without kick_start_time, no success. Then I run M106 S100 and M112. See the klippy.log....
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3827415/klippy.zip)

theophile, 2019-11-09:
Sounds like you need to add the line "max_power: 1" to your fan config.

Matrixer007, 2019-11-09:
@theophile 
Thank you so much, this is the solution. But I need a value of 0.01 to use the full range - I think because of the factor from 100- like written in the first post.
best regards
Uwe

grblmpfh, 2019-11-09:
I am not so sure that this is the solution. The power value should be between 0.0 and 1.0 and not between 0.0 and 0.01. There must be something wrong with the pwm-timer...

Matrixer007, 2019-11-10:
Hello!
After I did not let that rest, I tackled the matter with the multimeter.
If the factor is 1.0, M106 S255 will also be able to supply 24V at the connections. The regulation works so perfectly. But I installed an electric capacitor, which I forgot to mention, so that did not work with factor 1. Finally, the problem is found and with an Elko 100mF and the max_power: 0.01 the whole thing finally works.
Thanks for your help!

closed 2019-11-11 by Matrixer007