Klipper3d/klipper #2160: MAX31855 : Short to GND

smark-, 2019-11-08:
Hi @KevinOConnor and @cruwaller, I've been struggling for the past days to get the MAX31855 themocouple interface but I keep getting the following error: "MAX31855 : Short to GND"
I'm sure that the hardware is working because I have athe same interface board (made for both the MAX31855 and MAX6675) that has a MAX6675 and that one works correctly.
I've even tried to comment out the error check in the code, but the temperature reading is off if I compare it to the thermistor I'm using as reference. Any clues?
[klippy.log](https://github.com/KevinOConnor/klipper/files/3825286/klippy.log)

KevinOConnor, 2019-11-08:
You don't appear to be running code from this repo.  This github issue tracker is exclusively for code from this github repo.  You'll need to reproduce using the pristine code from this repo in order for us to assist.

-Kevin

KevinOConnor, 2019-11-09:
On a closer look, it doesn't seem the code you are running is much different from master.  It's still a good idea to reproduce using the pristine code.

That said, I don't know why you'd get bad values.  One thing you could try is a much slower spi rate (eg, 10000).  You might be getting noise on the spi lines.

-Kevin

smark-, 2019-11-09:
Kevin thanks for your support. I've switched to Klipper master, as you recommended, but I still haven't managed to make it work. I suspect that the 31855 was truly faulty, I measured one of the two terminals and it was indeed shorted to GND. Therefore I've grabbed an interface with the MAX31856 but I get another error: "Max31856: Cold Junction Range Fault". I've also lowered the spi bus speed to 100000, the minimum that you can set, and this hasn't solved the issue.
I've tested the same interface on an Arduino board and it works flawlessly. Has the MAX31856 code been tested as far as you know?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3827356/klippy.log)

KevinOConnor, 2019-11-10:
My understanding is that others have gotten the MAX31856 running.  However, there is a good chance that the sensor's config settings need to be set properly (see the details in the config/example-extras.cfg file).

-Kevin

closed 2019-12-13 by KevinOConnor