Klipper3d/klipper #2161: Use 2 drivers for same extrudeur

mrobotmaker, 2019-11-08:
I want to use 2 motors for my E0 : 

- One on the head's printer.
- One next to the filament spool. 

Can I declare in config file 2 outputs for step, dir and enable for E0 and use E1 physically pins to have the maximum torque and limit current inside 2 drivers? 

exemple:

`[extruder]`
`step_pin: ar26 and ar27
`

Thanks, 

Max

KevinOConnor, 2019-11-09:
Looks like a dup of #1267 .

-Kevin

dmbutyugin, 2019-11-12:
@mrobotmaker isn't your supposed setup dangerous though? The two extruders pushing the same line of filament will not be in perfect sync (because of slight differences in filament grip, etc.), so without a feedback loop one will be pushing slightly more filament than the other one. Best case, one motor will skip steps eventually, worst case the filament can break. My understanding is that multi-stepper extruder is best fitted for color/filament-mixing extruder setups, like diamond or cyclops.

KevinOConnor, 2019-11-15:
I'm going to close this in favour of the existing #1267 .

-Kevin

closed 2019-11-15 by KevinOConnor