Klipper3d/klipper #2166: Feature request: make Z velocity and acceleration limit independent of X and Y

Spacemarine2018, 2019-11-11:
At the moment max_z_velocity and max_z_accel can not be greater than max_velocity and max_accel, although I could not find any hint on this behaviour in the docs.

I don't think this limitation is neccessary for most printers today. Since I'm building a high speed coreXY pen-plotter, I ran into this limit, as lifting the pen (z-axis) has much less inertia than moving X and Y. 

One workaround would be to play with fake step distances, but it's not elegant. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3830703/klippy.log)

KevinOConnor, 2019-11-15:
A developer would need to implement this.  It would be similar to the work necessary to support independent x/y acceleration - #679 and #256 .

-Kevin

Spacemarine2018, 2019-11-16:
Maybe I'm woring, but I don't think any real development is neccessary. max_z_velocity and max_velocity can already be set independently. But as soon as  max_z_velocity > max_velocity an error is given. So it should be enough to just remove that check.

KevinOConnor, 2019-12-01:
I'm going to close this in favor of the existing similar issues on this (#679 and #256).  If a developer wishes to implement and submit then that's great.

-Kevin

closed 2019-12-01 by KevinOConnor