Klipper3d/klipper #2167: TUNING_TOWER and temperature

antst, 2019-11-11:
Maybe I am missing something, but do I understand that that TUNING_TOWER works only for extended commands which have single parameter? 
I want to use it for temperature calibration, but SET_HEATER_TEMPERATURE has  two parameters, and M104/M109 don't have parameters.

antst, 2019-11-11:
Just in case, when I do
`SET_HEATER_TEMPERATURE HEATER=extruder TARGET=XXX` it works

jperrin72, 2019-11-14:
>Just in case, when I do ...

@antst I got the same limitation with tuning tower feature. My workaround was to create a specific macro for each extruder that would get a single input parameter to set temperature.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

KevinOConnor, 2019-11-15:
The tuning tower command works for me.  You'll need to attach a log (and be sure to run M112 after the issue) in order for us to assist.

-Kevin

closed 2019-11-18 by klipper-gitissuebot