Klipper3d/klipper #2168: Duet Maestro Z-Axes only goes down

stefanfoulis, 2019-11-11:
I very well may have misconfigured something. Since Maestro support is relatively new I opted to create an Issue anyway.

For a _Duet 2 Maestro V1.0_ on an _Ender 3_ I followed [this wiring guide](https://duet3d.dozuki.com/Guide/Ender+3+Pro+and+Duet+Maestro+Guide+Part+1:+Wiring/37) and adapted [`generic-duet2-maestro.cfg`](https://github.com/KevinOConnor/klipper/blob/v0.8.0/config/generic-duet2-maestro.cfg) as the config for klipper.
I had to invert `dir_pin` to get `x` and `y` axes to go in the correct direction.

However  my z-axis is always going down, never up.

`G91`
`G1 Z10 F200` --> goes down
`G91`
`G1 Z-10 F200` --> goes down

This is my `stepper_z` config:
```
[stepper_z]
step_pin = PC28
dir_pin = PB4
enable_pin = !PA1, tmc2208_stepper_z:virtual_enable
step_distance = .0025
endstop_pin = probe:z_virtual_endstop
position_max = 250
position_min = -3

[tmc2208 stepper_z]
uart_pin = PA9
tx_pin = PA10
select_pins = !PC14, PC16, !PC17
sense_resistor = 0.075
microsteps = 16
run_current = 0.7
stealthchop_threshold = 30
```

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3833266/klippy.log.zip)

stefanfoulis, 2019-11-11:
Based on hints I got in the discord chat:

Between A1 and L1 there is 6Ω resistance. Between B1 and L2 there is 0Ω resistance. I don't know why the naming of the pins on the board for the Z connector is different than on X,Y and the extruders. :thinking:
I also checked the wire itself and all 4 connectors are fine.

I also tried turning the plug around... and then it only went up. So it seems wiring and stepper motor are fine.
Then I inverted dir_pin again... still went up.
So I am speculating that `PB4` might not be the right dir_pin after all :thinking:

stefanfoulis, 2019-11-11:
I connected the Z-Stepper to the E1 output on the board... and that worked fine... up and down.
So maybe some bug in klipper regarding the mapping of the pins for sam4s8c.
Or perhaps my board is broken.

KevinOConnor, 2019-11-12:
This was fixed a few days ago (commit f723a314) - you'll need to upgrade to the latest code.

-Kevin

stefanfoulis, 2019-11-12:
@KevinOConnor awesome, thanks! I can confirm it works for me on latest `master`. I'm glad my board is not broken.
And here I was, thinking: "I already have enough unknown moving parts. Let us use the stable release without all the experimental stuff".

closed 2019-11-12 by stefanfoulis